guysoft/OctoPi #308: Multiple Octopi Troubles

jedi5200, 2017-02-18:
Hi!
I've been having a lot of trouble lately. First, I flashed the stable Jessie image, but I couldn't even use startx (said the bash didn't recognize it), so I flashed the wheezy stable image. I finally got the wifi up after a long time by deleting the mac-made quotation marks around my ssid and psk in octopi-network.txt and replacing them with pi-made quotations. I puttied into my pi and got into the web interface. I tried to update it, but the update failed. my pi restarted for something and I haven't been able to get back to the web interface since. I thought it was the port number that was throwing me off, but I even edited config.yaml and set a port number, but that did nothing.
I also haven't been able to use my raspicam v2. MJPG streamer was already installed, so the
cd ~ sudo apt-get install subversion libjpeg8-dev imagemagick libav-tools cmake; git clone https://github.com/jacksonliam/mjpg-streamer.git; cd mjpg-streamer/mjpg-streamer-experimental; export LD_LIBRARY_PATH=.; make and
./mjpg_streamer -i "./input_raspicam.so -fps 5" -o "./output_http.so" did not work. it says there is no file called ./mjgp-streamer and the make command has no target. (semicolons separate commands.)
My octopi is ver. 0.12.0. I didn't realize I needed the octoprint version until after it was closed.
I have no printer, but one is coming soon and I am trying this in expectation. it will be an anet a8.
Thanks in advance!

guysoft, 2017-02-18:
1. There is no X in 0.13.0 and above, there is no need for it
2. Do you have a question? - I don't understand what exactly you have problems with now.

jedi5200, 2017-02-18:
what do you mean there is no X, and I want to connect to the web interface.

guysoft, 2017-02-18:
You connect to the web interface from another computer, not from the Pi, the Pi is using its CPU to control your printer (well, printer to be)

jedi5200, 2017-02-18:
right. ive been using my surface pro 4 to putty into my pi, then I put the ip address into edge. I am confused whether I need the port number or not.

jedi5200, 2017-02-18:
how do I update my octopi version?

ntoff, 2017-02-19:
>I puttied into my pi and got into the web interface. 

You don't get into the web interface through putty, putty is for console access to the underlying OS

>I tried to update it, but the update failed. my pi restarted for something and I haven't been able to get back to the web interface since. 

Update what? The underlying raspbian OS or octoprint? Are you talking about the popup in octoprint that prompts you to update or did you run a set of commands through putty for the update?

>I thought it was the port number that was throwing me off, but I even edited config.yaml and set a port number, but that did nothing.

It's extremely easy to mess up config.yaml so I suggest you don't touch it unless you A) really have to and B) know precisely what you're doing. When using octopi it also includes haproxy with forwards port 5000 to port 80 internally so when accessing the web interface make sure you aren't adding the :5000 at the end of the URL (that's only if you're running octoprint without haproxy)

>how do I update my octopi version?

Technically you don't, octopi is just the overall name of a raspberry pi image that contains octoprint and a precompiled mjpeg streamer, and haproxy.  Occasionally octo*print* will prompt you to update it, you should just be able to click the update button in the web interface.

You seem to be confusing the manual install instructions with the octopi image instructions. When you download octopi and flash it to a raspberry pi, that's all you need to do (aside from entering your wireless details). You don't need to follow any of the other instructions under the octoprint manual install wiki.

jedi5200, 2017-02-19:
I know I don't get into the web interface through putty. I have already connected to the web interface, I just cant reconnect. 

I tried to update octoprint through the web interface when it prompted me, but the update failed. 

I am going to try to connect again.

jedi5200, 2017-02-19:
does it matter that my ip address is a 10.0.1.xxx?

ntoff, 2017-02-19:
looks like an internal ip address, as long as that's what your local network uses then yes it's fine.

I suggest you start again from scratch, just to make sure we all know what's going on and that you have a fresh and unaltered install.

Download the octopi image, flash it to an SD card and if you can connect a keyboard and monitor directly to it for the initial setup and type sudo nano /boot/octopi-network.txt and alter the config that way, hit ctrl + x and follow the on screen prompts (y, then enter I believe).

Then on your computer or tablet or whatever, access the web interface and let octoprint prompt you for the update, click the update button. If it fails again, using the screen and keyboard connected directly to the pi, navigate to ~/.octoprint/logs/ and look through the last few lines to see if there's anything to indicate why it failed.

jedi5200, 2017-02-19:
which image is better: wheezy or Jessie lite?

ntoff, 2017-02-20:
This is a bug tracker, not a help forum. I'm sorry but I don't want to tie up the git issue tracker with this. Please go ask your question either on the 3dprinting subreddit, the raspberry pi subreddit or in the google+ group. Clearly your issue isn't one of buggy code but a misunderstanding of instructions which is something better discussed elsewhere.

google+ https://plus.google.com/communities/102771308349328485741

reddit: https://www.reddit.com/r/3Dprinting or https://www.reddit.com/r/raspberry_pi

closed 2017-03-14 by guysoft