guysoft/OctoPi #309: cannot conect to wifi

ARKopp, 2017-02-19:
I cannot get my wifi working and connect with ssh. 
the octopi-network.txt does not seem to work. Only when I type in everything in like this below I get an ip adress but then i cannot connect with ssh. 

sudo nano /etc/network/interfaces

# WLAN
allow-hotplug wlan0
iface wlan0 inet manual
wpa-ssid "WLAN-NAME"
wpa-psk "WLAN-PASSWORT"

guysoft, 2017-02-21:
What version of OctoPi are you using?

WinDroidGuy, 2017-03-18:
I'm using the latest on the site 0.13 with the same issue, if I connect via ethernet it works fine; but my modem/router are in a completely different part of the house for that to work.

shanehull, 2017-03-28:
Same issue here.

I'm using the latest image and have confirmed that it's not a text editor issue.

WinDroidGuy, 2017-03-28:
@shanehull which OS are you using to edit?

shanehull, 2017-03-28:
I'm using nano SSH'd into the Pi @WinDroidGuy.

shanehull, 2017-03-29:
Tried a fresh image and it just worked!

breadfan13, 2017-03-31:
Try this, then do the edit to your network interfaces you referenced the problem could be with how the " quotes are saving. or try using the @ sign if you don't want to change the locale

allow-hotplug wlan0
iface wlan0 inet manual
wpa-ssid @WLAN-NAME@
wpa-psk @WLAN-PASSWORT@

http://rohankapoor.com/2012/04/americanizing-the-raspberry-pi/

JCacho, 2017-05-24:
Has anyone discovered another solution? Still can't connect it.

shanehull, 2017-05-24:
Nope. Mine ended up connecting but never gets an IP. When you run iwconfig does the adapter say it's connected to your assigned SSID?

John-Mc, 2017-05-25:
@JCacho   -  If the iwconfig command does not show you are connected, the next thing to try is:
`sudo iwlist wlan0 scan`  
Buried in the resulting output, you should see the SSIDs of any WiFi networks that your Pi finds (even if it is not connected to them). This will tell us if your Pi can even see your network.

If it sees your network but is not connecting to it, one of the more common problems (especially if you are using Apple's TextEdit on a Mac to edit the connection information) is not having the correct type of quotation marks around your SSID and password. Take a close look at your octopi-network.txt file. If the quotation marks around your SSID and password are curly, that is a problem. If you see that, the quick fix is to copy and paste the straight type of quotation marks from another part of the document. (We can also tell you how to shut off TextEdit's quotation mark substitution behavior, if this was the problem.)

jormono, 2017-05-29:
Hello, new to github (and a complete novice when it comes to these types of things)here, I'm having a problem connecting to wifi as well. I've replaced the quotation marks from within the pi, and then rebooted. For some reason the scale of the screen seems to have changed and I couldn't see the results of the network scan, though I had done that before and saw my neighbors wifi network and wireless printer, but didn't see my own network (assumed it was off screen and couldn't figure out how to scroll and see it). I've tried every trick I know, at this point I'm wondering if it would be easier to start a fresh build from raspbian pixel and installing octoprint onto it. Any help is much appreciated!

idpage, 2017-06-21:
I was able to fix it by ignoring quotation marks but both my side and password were a single word with no whitespace characters

guysoft, 2017-07-19:
Closing looks fixed

closed 2017-07-19 by guysoft