guysoft/OctoPi #310: Octopi/whole raspberry pi zero just hangs.

famewolf, 2017-02-25:
This has happened several times while the machine was idle but it just died in the middle of a print.  I've used Linux for years and it's normally rock steady.   The only messages I can see are alerts in the console.  I've included two different "crashes".   Tell me what else you need and I'll attach it.

pi@octopi:~/.octoprint/logs $ 
Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80845.809634] Internal error: Oops: 17 [#1] ARM

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80845.941658] Process swapper (pid: 0, stack limit = 0xc07c8188)

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80845.949272] Stack: (0xc07c9df0 to 0xc07ca000)

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80845.955428] 9de0:                                     00000000 d2be43c0 c07c9e24 c07c9e08

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80845.967200] 9e00: c044d768 c00e2ce4 d6509300 d6509300 d646e400 00000005 c07c9e3c c07c9e28

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80845.978960] 9e20: c044d7f8 c044d700 00000001 d6509300 c07c9e54 c07c9e40 c044d960 c044d7d4

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80845.990748] 9e40: d6509300 d646e4dc c07c9e74 c07c9e58 c03a0ad0 c044d938 d646e520 00000000

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.002604] 9e60: c07d2a1c 00000000 c07c9e94 c07c9e78 c0027940 c03a0a74 c0842ab8 c081b9b8

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.014612] 9e80: 00000040 00000006 c07c9edc c07c9e98 c0026f00 c00278c8 00004987 c07d7c9c

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.026803] 9ea0: 00000001 00200000 007ae707 00000009 c07c9ecc 00000000 c081b9b8 00000000

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.039145] 9ec0: 00000000 00000000 c083c5e5 c083c5e5 c07c9ef4 c07c9ee0 c0027450 c0026ddc

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.051662] 9ee0: c005df80 00000000 c07c9f1c c07c9ef8 c005ac70 c0027390 c07c9f30 c00107f4

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.064263] 9f00: 60000013 f200b200 c07c9f64 c083d048 c07c9f2c c07c9f20 c000935c c005abf8

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.076982] 9f20: c07c9f84 c07c9f30 c0538118 c000933c 00000000 00000000 00000000 c07caf40

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.089822] 9f40: c0842350 c07c8000 c07ca090 00000000 c083d048 c083c5e5 c083c5e5 c07c9f84

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.102710] 9f60: c07c9f78 c07c9f78 c00107f0 c00107f4 60000013 ffffffff c07c9f9c c07c9f88

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.115590] 9f80: c004e898 c00107d0 c053d17c c07ca000 c07c9fac c07c9fa0 c0531e68 c004e750

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.128478] 9fa0: c07c9ff4 c07c9fb0 c0771c88 c0531e08 ffffffff ffffffff c07716d0 00000000

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.141362] 9fc0: 00000000 c07ac528 00000000 c0842494 c07ca024 c07ac524 c07cf5a0 00004008

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.154239] 9fe0: 410fb767 007aba18 00000000 c07c9ff8 00008070 c0771948 00000000 00000000

Message from syslogd@octopi at Feb 24 20:17:46 ...
 kernel:[80846.373274] Code: e92dd830 e24cb004 e52de004 e8bd4000 (e5903000) 
Write failed: Broken pipe

--------------------
Middle of a print job
--------------------

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6643.987937] Internal error: Oops: 81f [#1] ARM

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.123388] Process octoprint (pid: 736, stack limit = 0xd663e188)

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.131441] Stack: (0xd663fe70 to 0xd6640000)

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.137656] fe60:                                     00000000 d17fe8e0 d663fea4 d663fe88

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.149485] fe80: c044d768 c00e2ce4 d6481600 d6481600 d647ec00 00000005 d663febc d663fea8

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.161294] fea0: c044d7f8 c044d700 00000001 d6481600 d663fed4 d663fec0 c044d960 c044d7d4

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.173121] fec0: d6481600 d647ecdc d663fef4 d663fed8 c03a0ad0 c044d938 d647ed20 00000000

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.185006] fee0: c07d2a1c 00000000 d663ff14 d663fef8 c0027940 c03a0a74 c0842ab8 c081b9b8

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.197040] ff00: 00000040 00000006 d663ff5c d663ff18 c0026f00 c00278c8 00000000 c07d7c9c

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.209262] ff20: 00000001 00400040 0009ae04 00000009 d663ff4c 00000000 c081b9b8 00000000

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.221641] ff40: 00000000 00000000 008ebea8 002e8758 d663ff74 d663ff60 c0027450 c0026ddc

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.234191] ff60: c005df80 00000000 d663ff9c d663ff78 c005ac70 c0027390 d663ffb0 00083d58

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.246842] ff80: 80000010 f200b200 00c5387d 00c5387d d663ffac d663ffa0 c000935c c005abf8

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.259588] ffa0: 00000000 d663ffb0 c0538428 c000933c 00000000 acf94b20 0032dc44 0000001d

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.272452] ffc0: b6472746 b6ccd080 003349d4 0012bf04 00000011 008ebea8 002e8758 af0f487c

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.285356] ffe0: 0000001e b30fde98 0012bfbc 00083d58 80000010 ffffffff 00000000 00000000

Message from syslogd@octopi at Feb 25 00:24:30 ...
 kernel:[ 6644.456049] Code: e2803010 f5d3f000 e1935f9f e2455001 (e1832f95)

guysoft, 2017-02-27:
I have no hardware to test here (no Pi zero).
The broken pipe errors might indicate you ran out of disk space, I'd check that.
Apart from that, I have no way to reproduce.
@foosel ?

foosel, 2017-02-27:
Looks like a kernel oops. So way below anything that OctoPrint is responsible for. I suggest taking a look into the syslog, might have more information. Possibly a misbehaving driver, eg of the used network dongle. But that's all I can tell you, haven't heard of similar issues from others that use OctoPi with a zero. Personally I also wouldn't do it, way too much hassle with all those adaptors necessary.

famewolf, 2017-02-27:
Thanks.  Looks like the problem was the usb hub with built in ethernet (I
replaced it with a standard 7 port powered hub)  as when I disconnected it
and switched to wifi the problem appears to gone away.   Of course I
realized that after I'd taken another microsd and setup Raspian with
octoprint, mjpg-streamer etc manually.

It's not all that big a deal actually...add 1 7 port self powered hub and
and OTG cable..the pi zero gets it's power from the hub...no 2nd cable
needed...no hdmi needed although I have the adapter.  Everything else
(wireless keyboard/trackpad, wifi dongle, usb webcam etc) plugs into the
hub as normal.

Thanks for the followup.  This can be closed.

closed 2017-03-12 by guysoft