guysoft/OctoPi #318: OctoPi 0.13 does not work with Pi Zero W

mitsumaui, 2017-03-04:
When installing OctoPi 0.13 to SD card, no Wifi adapter is detected on the Pi Zero W. This is built in now, so expect an updated Raspbian image is required? Will test further with Raspbian.

guysoft, 2017-03-04:
1. It will not work because the Pi Zero W was released after 0.13.
2. The raspberrypi themselves didn't get it working yet. If you look at the
release notes ( http://downloads.raspberrypi.org/raspbian/release_notes.txt
) you will see it says "initial support for Pi Zero W". Aka its still
initial and we are expcting them to release a version with fixes.

I will get the nightly coming out hopefully tomorrow, the wifi in my house
has been acting up and breaking the build server.

mitsumaui, 2017-03-05:
Thanks for the response! I saw this in the Raspbian rel notes so assume it's mostly ready to go:

2017-03-02:
  * Updated kernel and firmware (final Pi Zero W support)

I've just installed octoprint on the latest Raspbian Lite image manually and will do some testing! Once you have this into nightlies - let me know (along with a brief instruction on how I can load to SD card for testing) and I can do some testing.

guysoft, 2017-03-07:
Huh, they changed it. It said "initial" a few days ago.
Oh well, @foosel already got errors when trying the camera. And we are on the move for 0.14.0 RC, when we are ready.

foosel, 2017-03-07:
Some more infos on the errors I encountered...

Utilizing actual bandwidth on the WiFi interface (as in, streaming from the webcam in the "Control" tab in OctoPrint and also directly via `/webcam/` plus doing a simple speed test for downloading something via `wget`) I saw huge CPU consumption by `kworker` and `irq` in the output of `top`. And printing a tri spinner (lots of curves) and trying to watch the webcam resulted in endless stuttering thanks to the above mentioned processes eating up ~50% of all available CPU resources.

That was still against the "initial support" image, aka 2017-02-16 though. I did a manual `rpi-update` on that which pushed the kernel version to 4.9 and reduced the CPU load during WiFi usage to something around 25%. Still some stuttering but at least better.

I then installed the SnapStream plugin and used that configured for 4fps instead of the actual webcam stream. That brought the CPU usage down to 15% and the print went through. 

Still... a decidedly "meh" experience so far. As a side note, I also tested against a Pi3 then out of curiousity: same behaviour but doesn't matter thanks to four cores being available.

tldr: Actually using the WiFi of the RPi Zero W to transfer data (= watching a webcam stream of a print) eats a lot of CPU and can severly impact achievable print quality.

foosel, 2017-03-07:
Some quick shots:

[image]

This is only one browser connected to `http://octopi.local/webcam/stream.html`. OctoPrint server isn't even running.

[image]

And this is a simple `wget -O /dev/null http://somehost:someport/somelargefile.gcode` executed on the Pi. 

And look at that load average. Hilarious. As a reminder, we are not doing anything computational intensive here. Just pumping some data through the wifi interface.

foosel, 2017-03-07:
@nophead I think I already mentioned that once, but I never had such issues with USB wifi dongles. This mmc stuff though... 

Now testing one of my dongles with the RPi Zero W.

foosel, 2017-03-07:
@nophead *now* I finally see what you mean with CPU load by `haproxy` ;)

[image]

That's a running print with the Control tab (and hence the webcam stream) open. USB dongle, internal NIC is disabled. Slight stuttering but only at the start. Couldn't let it run (appointment) so no idea how that would develop over time. Next to no lag though (same setup with internal nic had lots of lag).

[image]

That is only streaming from `/webcam/`, OctoPrint server is disabled. USB nic, internal one disabled.

[image]

And `wget`.

Needs more tests but current gut feeling based on observed print behaviour: USB WiFi dongle performs better than built in interface.

guysoft, 2017-03-11:
Next nightly would be with the latest base image

pposthoorn, 2017-03-14:
Newest build works for me with the new raspberry Pi zero W with Pi camera.

calonmerc, 2017-03-17:
Any additional testing needed on this? I've got a Pi Zero W doing nothing (but no Pi cam, still in the mail).

foosel, 2017-03-17:
More testing is always good!

@guysoft i released OctoPrint 1.3.2 yesterday, so that is now ready for inclusion in 0.14 as well (and might have already made it into this night's nightly? Not sure if that is build against master, would make sense though)

guysoft, 2017-03-17:
nightly build is against master. If you got it in before midnight UTC it
should be there

foosel, 2017-03-17:
Ah, great. I wasn't sure anymore if the nightlies were master or devel branch. Release was 3pm CET/2pm UTC so that should have made it in :)

robomaniac, 2017-03-19:
hey guys,

it's my first time use of the octopi. I am using a Zero W and downloaded this nightly build 
`2017-03-17_2017-03-02-octopi-jessie-lite-0.14.0.zip	2017-03-17 01:11:01 UTC`

and change the wifi settings in the .txt file and boom got everything working! I did my first print too.
My camera was a black screen and did not undestand why until I went in the basement to check my printer and saw that the camera felt face down on the table! hahaha!!

thanks!
I just wanted to tell you that I got it working on the Zero W since I saw some people having difficulty.
I also tried the 0.13 until I came in this thread and tried the nightly build.

my questions are: When do I know octoprint release a stable version for the Zero W?
I can't find the release #
here? [http://gnethomelinux.com/OctoPi/nightly/#rpi-stable](url)

if I want to update my nightly once in a while, I need to use these scripts?
[https://github.com/guysoft/OctoPi/tree/devel/src/nightly_build_scripts](url)
basically I need to use github commands to find that folder and execute those script?

thanks

guysoft, 2017-03-20:
@robomaniac Hey,
So
1. Soon, and it would be mostly identical to the current nightly, so you can use that quite surely
2. OctoPrint update itself automatically, you will get notified in the browser. don't confuse OctoPrint with OctoPi. Those scripts are used toto build the image itself

crazyquark, 2017-03-29:
I confirm that debian jessie lite 2017-03-02 does run on pi zero w.

bolsoncerrado, 2017-05-05:
Hey @robomaniac so are you able to run octopi on the ZeroW, access remotely and watch the camera without much hassle? I don't need 60fps video feed, just quick checks to see steady temps and no spaguetti incidents should be enough ;)

wdy say? :)

robomaniac, 2017-05-06:
yes sir, It's been running since Mars. I had weird bug that I can not reach
the Zero via my laptop but works with my ipad and iphone. It's been stable
for a few weeks now and that bug did not appear again.

there is lots of great plug-in that makes the experience even better!

Go for it!

bolsoncerrado, 2017-05-07:
Great! Thanks for letting me know.

guysoft, 2017-07-23:
Closing, 0.14 released

closed 2017-07-23 by guysoft

jalopezsuarez, 2018-01-30:
@foosel did you tried Pizero with USB dongle? Same performance problems ?

foosel, 2018-02-07:
@jalopezsuarez see [this comment](https://github.com/guysoft/OctoPi/issues/318#issuecomment-284782199). TLDR: No. Dongles perform way better. Kinda defeats the purpose though, especially on something as IO limited as the 0w. Use a Pi3.

EpicLPer, 2018-02-08:
Uhm... Is this "solved" now since it's closed? Can you use the Zero W now with the latest 0.14 release?

Penfold81, 2018-02-08:
Works fine on all mine?

Thanks

Dave

iaglet, 2018-03-14:
Is this still an issue? I have a Pi Zero W running OctoPrint without issue. It can be a little slow to boot but it works. Now I am worried cause I want to rebuild the darn thing.

guysoft, 2018-03-14:
@iaglet the issue is closed

ettiennegous, 2018-05-08:
Can anyone confirm or disprove that Raspberry Pi Zero W is powerful enough if you run a Raspberry Pi Camera but not good enough if you run a USB web cam?

I see in this thread some responses where people say its fine and other complain.
Based on previous reading the Raspberry Pi Cam decoding is done right on the GPU vs USB cam will load up the CPU hence the huge performance hit some users are seeing.
Link below where a similar discussion happened.
https://www.raspberrypi.org/forums/viewtopic.php?p=607023

foosel, 2018-05-08:
[The Wifi on the Zero W gobbles up CPU cycles for breakfast when bandwidth is utilized](https://github.com/guysoft/OctoPi/issues/318#issuecomment-284762963). Webcams utilize bandwidth. Don't use the Zero W for OctoPrint, especially if you plan on using a webcam with it.

fornellas, 2018-05-08:
I have been using for months, including watching video stream while
printing. Although not the fastest, it has proven to be extremely reliable.

foosel, 2018-05-08:
I knew someone would chime in and say "It works perfectly for me!!1!". Be happy if it does. It certainly didn't in all tests that I did and I also got a lot of confirmation from other people in the community that it simply grinds to a halt. Anyone deciding on using that device, be my guest, but don't be surprised when you run into (predictable) performance issues left and right.

iaglet, 2018-05-08:
So some behavior is making sense. I haven't had any full fledged failures but some of the strange behaviors make more sense.  While I can't say that it is related, it is horribly slow rebooting.

fornellas, 2018-05-09:
The board is not fast by nature... So, things are usually quite slow

iLLiac4, 2018-05-22:
In latest version booting is slow as hell (not usable). Using older version boots much faster with Zero W.

guysoft, 2018-05-22:
@iLLiac4 what version?  Please open a seprate issue withthe full details

jalopezsuarez, 2018-05-22:
Not recommended install on PiZeroW, because of the performance. Anyway you should do a clean fresh install and try again. Please create a new post for details.

iaglet, 2018-05-25:
Even though I have a PiZeroW running Octopi and had success with it, I am now realizing that I was still having performance issues and would recommend against it. I would not do that over again. It's too bad.. it's a very elegant form factor but all the issues being described, I observed.

DarthKegRaider, 2018-07-26:
I too have been using my spare Pi0W (heatsinked) driving my Rostock Delta almost perfectly for approx 6 months.  Mainly due to the tiny form factor, but will be migrating to the Pi3B, once I design a housing for it to replace one of the current side covers, thus integrating my pi into the printer.

I have only had one (1) serious lockup using the zero, as it stays online for months at a time.  Rebooting is painfully slow though, taking approx 6 minutes before the webpage is available.  You do notice the printer slow down and get 'jittery' if you upload files with the webcam running.  That said, if they ever release a quad-core Zero, I think I'll put it back.  I like the size!

julcoh, 2018-08-28:
@mitsumaui Did you try the Banana Pi? Interested in getting a webcam working with this method as well.

Cova, 2018-08-29:
So reading through this whole thread - it sounds safe to assume that a zero-w running octoprint will be fine for me since I have no intention of connecting any kind of camera to it - just don't upload new gcode files (use significantly bandwidth) while a print is in progress.

DarthKegRaider, 2018-08-29:
@Cova , Give it a go if you have one laying around.  It might work fine for you, your SD card might be faster than mine :)

tparenti, 2019-12-18:
For anyone finding this page through the octopi warning - I have been using Octopi v0.15.1 successfully for about a year now, I do have the 1080p camera hooked up to it, streams work pretty well, the cpu does spike to 75% while printing and watching a stream though. im on wifi, not even that strong of a connection. Printer is a prusa i3, great quality now with several printer mods. CPU usage hovers around 15% on idle and 25-30% average during printing without watching the camera.

HorstBaerbel, 2020-01-20:
Maybe H.264 encoding instead of MJPEG might help, as the bandwidth requirements are lower. The Pis can also hardware-encode H.264, so the CPU utilization should be lower too. [I tested this](http://lektiondestages.blogspot.com/2016/12/stream-hardware-encoded-h264-video-from.html) a long time ago, maybe someone is willing to make this work in OctoPi...

estcap2, 2020-08-08:
I see a lot of people complaining that using the Pi Zero W + Camera (either Pi Cam or USB Camera) is not recommended. But what about using the Pi Zero W just to for example replace an unreliable SD Card slot? Is it still bad even for just printing without straming video?

iaglet, 2020-08-08:
I believe, though admittedly I could be explaining this incorrectly, the issue stems from how the Wi-Fi function is driven by the CPU. Functions that would typically be offloaded from that CPU, or run on this card. So when you have high throughput streaming, such as with a WebCam, it affects performanceYou potentially have issues with the serial communication.

-Stephen on Harmon

mitsumaui, 2020-08-08:
> But what about using the Pi Zero W just to replace an unreliable SD Card slot?

I ran a Pi Zero W with newer OctoPrint releases just fine up till about 6 months ago. It was fine for printing providing you don’t do too much on the Web Interface while it’s printing, i.e install plugins etc. as that will hang up printing. I never noticed any pauses with 115200 baud and no webcam. I could use the ‘layer progress’ tab comfortably without maxing the CPU, but that was as far as I could push it. 

I switched out to an Orange Pi Zero as wanted to use a webcam which is fine aside having to install OctoPrint manually vs. just flashing an image.

Cjkeenan, 2020-12-07:
@tparenti 
> I do have the 1080p camera hooked up to it, streams work pretty well

Do you use a Pi camera connected to the camera port or just a USB Webcam? Also do you use a external USB WiFi adapter or just the on-board one?

lukeski14, 2021-01-05:
Wondering if using ethernet instead of wifi on the Zero W via this [ethernet hat](https://www.amazon.com/Ethernet-USB-HUB-HAT-Compatible/dp/B07T16RSFM/) would lighten up the load on the CPU. I've got both of these lying around, so I may attempt to try it (along with a Pi Cam).

DarthKegRaider, 2021-01-05:
> Wondering if using ethernet instead of wifi on the Zero W via this [ethernet hat](https://www.amazon.com/Ethernet-USB-HUB-HAT-Compatible/dp/B07T16RSFM/) would lighten up the load on the CPU. I've got both of these lying around, so I may attempt to try it (along with a Pi Cam).

After reading about this HAT, I would say not.  It gets all it's functionality from the micro USB, not the GPIO pins, so essentially be the same as plugging in a USB hub and USB an dongle.