guysoft/OctoPi #325: Video stream working but stops updating

al-jimenez, 2017-03-23:
This may just be a settings issue, but I have searched and can't find anything on it...

I am printing with OctoPi (on Raspberry Pi 3) and monitoring the print, and the webcam stream will disappear and be replaced with a [?] icon.    This happens with Safari almost right away and with Chrome after about 1 or 2 hours of printing.    If I refresh OctoPrint in my browser video comes back but the issue returns.  Maybe its a memory buffer setting thing???  Not sure.  Its annoying and pls, pls, point me in the right direction if I need to tweak things.    I am using a Logitech USB Webcam on the Raspberry Pi which works great but for this issue.

nophead, 2017-03-23:
Have a look at /var/log/messages. My guess is your camera stream stops and the mjpg_streamer restarts and produces a new stream that takes a browser refresh to be displayed.

My Logitech USB camera drops it connection occasionally. I have no idea why.

al-jimenez, 2017-03-23:
@nophead (Chris):  I found this link that may help...  I am working on implementing this in case this helps you with Logitech camera drops...  [https://github.com/foosel/OctoPrint/issues/441](url)

guysoft, 2017-03-26:
That url is broken, there is no issue 441

al-jimenez, 2017-03-26:
Its a closed issue, but I don't understand why the link is not working,  my saved link which looks identical goes there...   https://github.com/foosel/OctoPrint/issues/441

closed 2017-11-01 by guysoft