guysoft/OctoPi #330: Spent hours trying to get this working with no luck, please help!

CorrieDivitt, 2017-03-29:
I'm using the newest version of the Raspberry Pi 3 with built in wifi.

Using Windows 10 I installed OctoPi version 0.13 to the SD card.

I went into the octopi-network.txt and edited it as all of the guides explain but I cannot get it to connect to wifi. I made sure not to leave the quotation marks alone and only edit the SSID, password and remove the hash marks.

When I type iwconfig it says 

wlan0       IEEE 902.11bgn   ESSID: off/any
mode: managed access point: not-associate tx-power=1496 dBm

under lo and eth0 it says no wireless extensions.

If I type ifconfig I get some random hex looking characters where the IP should show.

Can you please give me some hints as to where to start off troubleshooting this? I've tried several manual workarounds guides and nothing is working.

markwal, 2017-03-29:
You may want to post this question on the google group: https://groups.google.com/group/octoprint or to the Google+ community: https://plus.google.com/communities/102771308349328485741.  There are more people monitoring those locations than here which is for problems with the OctoPi image or the script that creates it.

BTW IPv6 address do have a bunch of hex looking characters in them.

breadfan13, 2017-03-31:
Try this, 

Use the ethernet cable to connect to rpi3 via SSH. Then go to raspi-config 

Change your locale from great brittain to US

Save-->reboot

Now ssh back into your pi and manually configure the /dev/wpa_supplicant/wpa_supplicant.conf
add these lines

network={
  ssid="testing"
  psk="testingPassword"
}

Reboot and ENJOY! 

you might also try editing the octopi-network.txt file and replace the " quotes with the @ at sign.  I was able to get a pi running once using the @ sign without changing the locale.

breadfan13, 2017-03-31:
P.S. Little Background:

(Assuming you are in the US)

Rasbian is natively configured to us Great Britain as the locale. removing the GB locale and changing it to en_US.UTF-8 has worked for me every time. YMMV

The octopi text file has never worked for me. I always have to manually set up wifi

Here is a guide with pictures:

Americanizing your Pi
http://rohankapoor.com/2012/04/americanizing-the-raspberry-pi/

CorrieDivitt, 2017-03-31:
Thanks for the help. I managed to get it working after doing a reset of my router and a fresh install of octoprint. I also realized octoprint.local was not working and I had to use the direct IP.

markwal, 2017-04-01:
The name octopi.local (or whatever you have set as the host name, but octopi by default) is broadcast using the Bonjour protocol by the bundled Discovery plugin.  Native Windows doesn't recognize that protocol, but if you install Bonjour Print Services for Windows, it will: https://support.apple.com/kb/DL999

closed 2017-07-23 by guysoft