guysoft/OctoPi #332: OctoPi 0.14.0 RC1 status

guysoft, 2017-03-31:
Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/2017-03-31_2017-03-02-octopi-jessie-lite-0.14.0.zip
md5:
https://storage.googleapis.com/octoprint/2017-03-31_2017-03-02-octopi-jessie-lite-0.14.0.zip.md5

It is built against Raspbian 2017-03-02 ([Release Notes](http://downloads.raspberrypi.org/raspbian/release_notes.txt))

### Changes in the image

  * Added note to `/root/bin/webcamd` to use `octopi.txt` for configuration
  * Added note to `octopi-network.txt` for Mac OS X users to not use Textedit/properly configure Textedit
  * Added note with further instructions on usage to boot and login screen
  * Allow configuration of mulitple wifi networks via `octopi-wpa-supplicant.txt`
  * Better "service not running" error pages
  * Enabled SSH by default (we usually run headless, disabling it is not an option)
  * Updated stock `config.yaml` and init script for OctoPrint to 1.3.0+
  * Updated `install-desktop` script to only pull in needed packages
  * Fixes:
    * Prevent weird SSH issues on RPi3 ([#294](https://github.com/guysoft/OctoPi/issues/294))
    * Prevent duplicate X-Scheme headers in haproxy ([#239](https://github.com/guysoft/OctoPi/issues/239))
    * Workaround for an issue with recent `pip` versions and pyasn1 ([#276](https://github.com/guysoft/OctoPi/issues/276))

dbvandy, 2017-04-02:
After much beating of my head with the new ZeroW and the 13 build... I have Octopi up and running with this RC1... sending job to the delta now!

Thanks!

dbvandy, 2017-04-02:
welllllll.......

Lots of buffering and a fairly simple job has crashed 3 times...  How can I document what is going on for you?

seem to run better if you are not viewing the web server... idk.

dbvandy, 2017-04-02:
Send: N9301 G0 F6000 X18.241 Y6.083*64
Recv: skip 9310
Recv: ok
Send: N9302 G1 F2400 X18.847 Y5.477 E543.32087*5
Recv: skip 9311
Recv: ok
Send: N9303 G0 F6000 X18.241 Y4.951*70
Recv: ok 9287
Send: N9304 G1 F2400 X18.847 Y4.345 E543.34937*0
Recv: ok 9288
Send: N9305 G0 F6000 X18.241 Y3.820*64
Recv: ok 9289
Send: N9306 G1 F2400 X18.847 Y3.214 E543.37788*9
Recv: ok 9290
Send: N9307 G0 F6000 X18.241 Y2.688*79
Recv: ok 9291
Send: N9308 G1 F2400 X18.847 Y2.083 E543.40636*14
Recv: ok 9292
Send: N9309 G0 F6000 X18.241 Y1.557*67
Recv: ok 9293
Send: N9310 G1 F2400 X18.847 Y0.951 E543.43486*9
Recv: ok 9294
Send: N9311 G0 F6000 X18.241 Y0.425*79
Recv: ok 9295
Send: N9312 G1 F2400 X18.847 Y-0.179 E543.46332*41
Recv: ok 9296
Send: N9313 G0 F6000 X18.241 Y-0.705*97
Recv: ok 9297
Send: N9314 G1 F2400 X18.847 Y-1.310 E543.49180*39
Recv: ok 9298
[...]
Recv: Error:Wrong checksum
Recv: 
Recv: Resend:9299
Recv: ok
Send: N9299 G0 F6000 X18.241 Y7.214*77
Recv: skip 9308
Recv: ok
Send: N9300 G1 F2400 X18.847 Y6.608 E543.29236*12
Recv: Error:Wrong checksum
Recv: 
Recv: Resend:9299
Recv: ok
Send: N9301 G0 F6000 X18.241 Y6.083*64
Recv: skip 9315
Recv: ok
Send: N9302 G1 F2400 X18.847 Y5.477 E543.32087*5
Recv: ok 9299
Send: N9303 G0 F6000 X18.241 Y4.951*70
Recv: ok 9300
Send: N9304 G1 F2400 X18.847 Y4.345 E543.34937*0
Recv: ok 9301
Send: N9305 G0 F6000 X18.241 Y3.820*64
Recv: ok 9302
Send: N9306 G1 F2400 X18.847 Y3.214 E543.37788*9
Recv: ok 9303
Send: N9307 G0 F6000 X18.241 Y2.688*79
Recv: ok 9304
Send: N9308 G1 F2400 X18.847 Y2.083 E543.40636*14
Recv: Error:Wrong checksum
Recv: 
Recv: Resend:9305
Recv: ok
Send: N9305 G0 F6000 X18.241 Y3.820*64
Recv: skip 9307
Recv: ok
Send: N9306 G1 F2400 X18.847 Y3.214 E543.37788*9
Recv: skip 9308
Recv: ok
Send: N9307 G0 F6000 X18.241 Y2.688*79
Recv: skip 9306
Recv: ok
Send: N9308 G1 F2400 X18.847 Y2.083 E543.40636*14
Recv: skip 9307
Recv: ok
Send: N9309 G0 F6000 X18.241 Y1.557*67
Recv: skip 9308
Recv: ok
Send: N9310 G1 F2400 X18.847 Y0.951 E543.43486*9
Recv: skip 9309
Recv: ok
Send: N9311 G0 F6000 X18.241 Y0.425*79
Recv: skip 9310
Recv: ok
Send: N9312 G1 F2400 X18.847 Y-0.179 E543.46332*41
Recv: skip 9311
--- too many lines to buffer, cut off ---

dbvandy, 2017-04-02:
[Laser_blast_shield_1_0_a.zip](https://github.com/guysoft/OctoPi/files/888166/Laser_blast_shield_1_0_a.zip)

dbvandy, 2017-04-02:
I have disconnect mouse and keyboard from Pi Zero W and all seems well...  printing with no buffering now.

guysoft, 2017-04-02:
@dbvandy Ok please dont spam the thread, If you have new information please edit an existing thread and type "update:" with the new information you want to add.
I am editing your messages to make them readable.

guysoft, 2017-04-03:
@foosel Is this a no-go for the feature?
https://github.com/guysoft/OctoPi/issues/336

foosel, 2017-04-03:
@dbvandy What kind of mouse and keyboard and what kind of printer? I have not seen this behaviour in my own tests. Power issue? What does "Print jobs crash" mean, what happens exactly?

@guysoft #336 at least sounds like we need some kind of workaround for the current `raspi-config` behaviour, otherwise IMHO no-go. Will comment there.

dbvandy, 2017-04-03:
Standard usb keyboard, Logitech m510 wireless mouse, delta kossel 180x280
printer running marlin.

The 16 command buffer ran down frequently and then finally ran out causing
the printer to quit.

After removing the keyboard and mouse from the passive usb hub, the printer
did not hit the buffer much if at all.

I would assume that the limited memory and processing power of the zero w
cause the processor to get overwhelmed with the multiple usb devices.  I am
not aware of a way to monitor the cpu processes on the pi to verify that it
is bogging.

I can reproduce this effect by reconnecting the keyboard and mouse to the
hub.

Doug Vanderbilt
Senior Trainer / Consultant
Knowledge Tree Consulting
cell: 404.630.9840 <(404)%20630-9840>

foosel, 2017-04-03:
Looks like some serious communication issues causing the stuttering, printer keeps requesting lines that look like they are sent ok. So either they become garbage on the line or when they arrive at the printer. So probably less of a CPU issue and more of a power issue (make sure the power supply provides 5V, 2.5A), or an overloaded bus maybe. Considering it's a wireless mouse, I wouldn't be too surprised actually if it's power and the dongle of the mouse is demanding too much, causing brown outs. Try leaving the mouse out of the equation. Currently running a test print against the image on a ZeroW, have a wireless keyboard-mouse-combo AND a mouse connected, as well as a pi camera and the printer itself. No communication errors. Power supply is [this one](https://www.amazon.de/gp/product/B01FOTVJJC/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1), 5V, 2.5A.

> I am not aware of a way to monitor the cpu processes on the pi

I'd recommend ``top``, it's is already installed.

---

Side note: I'm still seeing the abysmal performance with regards to CPU load when the webcam tab is opened as already mentioned in #318. @guysoft Should we maybe add a "but not recommended" to the bit about support for the Zero W? I really have a bad feeling recommending this thing for running prints when the built-in wifi is used. Even a timelapse download could nuke prints with this kind of performance. Currently seeing 25% for `kworker` and 18% for `irq/86-mmc1` (wifi handling) with traffic, and THAT certainly will cause stuttering.

John-Mc, 2017-04-03:
I hope Guysoft's question "Is that a no-go for the feature?" #336 does not result in removal of the octopi-wpa-supplicant.txt feature. That feature is a major plus for those of us who need a pi to run on multiple networks and need to add new networks easily.

foosel, 2017-04-03:
If we can't find a solution for the raspi-config issue though we'll at least need a big fat disclaimer in that file or we'll never hear the end of this issue (endless stream of tickets).

foosel, 2017-04-13:
Considering [this](https://nvd.nist.gov/vuln/detail/CVE-2016-10229) and

```
OctoPi version    : 0.14.0
------------------------------------------------------------------------------

pi@octopi:~ $ uname -a
Linux octopi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
```

I guess we should also make sure that RC2 has a newer kernel - if one is available.

guysoft, 2017-04-13:
Is there a way to check if one is available?

foosel, 2017-04-13:
https://www.raspberrypi.org/documentation/linux/kernel/updating.md

Tried both, official package is still <4.5, what you get via rpi-update is 4.9 but not recommended according to the link above.

Let's see what will be the state after the weekend.

foosel, 2017-04-18:
New image is out, short summary of findings:

  * kernel still at 4.4.50 from Feb 20 but taking a look at the `4.4.y` branch on raspberrypi/linux it looks like the above CVE should be fixed in that one after all
  * `raspi-config` still the old version without the fix by @kennethjiang mentioned in #336, so we still need the workaround provided in #337
  * `libpam-chksshpwd` no longer missing, meaning users logging in via SSH and locally will see a warning about changing their password if they haven't changed it from the default `raspberry`

I suggest building RC2 against that new image.

guysoft, 2017-04-22:
@foosel Nightles are back, with the new image:

https://storage.googleapis.com/octoprint/2017-04-22_2017-04-10-octopi-jessie-lite-0.14.0.zip
MD5 - 0bb3aa2033e09fb0e59764fdb57f9d8b

Should we make RC2 out of this?

John-Mc, 2017-04-22:
When RC2 comes out, is it possible to update to this from within OctoPrint, or do I need to start fresh from that image? (I do have my OctoPrint update channel set to Devel RC).

guysoft, 2017-04-22:
@John-Mc Fresh image. You can move most of your OctoPrint preferences between images.

foosel, 2017-04-27:
@guysoft don't see why not. Also sorry, currently out of commission (wisdom teeth extraction + cold or something, lovely combination), hence the delay in replying to this.

shane84948, 2017-04-29:
Hello, Im having an issue getting the PI v2 camera to work with this build. Not sure if its something Im missing or another issue all together. I have already tried the get update and get upgrade and my GPU memory is over the 128 min. It is a new camera and cable (which is installed correctly) but I believe its has to be a setting somewhere im missing. Any help would be greatly appreciated. 

pi@octopi:~` $ raspistill -v -o test.jpg
raspistill Camera App v1.3.11
Width 3280, Height 2464, quality 85, filename test.jpg
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture
Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware ``updates

nophead, 2017-04-29:
I think you will always get that error with raspistill on OctoPi because the camera will be already used by  mjpg-streamer.

dookie2000ca, 2017-04-29:
The release [from a week ago](https://github.com/guysoft/OctoPi/issues/332#issuecomment-296361841) is working great on my RPi 3 with my D300VS printer and the obligatory Logitech C270.  
Wireless performance is much improved.

John-Mc, 2017-06-02:
Did an OctoPi 0.14rc2 ever come out? I thought I had seen mention of it, but can't find it - and the link to the nightly builds is not working for me.

Penfold81, 2017-06-02:
You could always force the update in advanced settings

John-Mc, 2017-06-02:
I could force an OctoPrint update. Not sure you can do an OctoPi update that way. Pretty sure you have to install that from the new image.

guysoft, 2017-06-18:
Please use RC2 at #367

closed 2017-06-18 by guysoft