guysoft/OctoPi #333: OctoPi  public wifi log-in

prangsoa, 2017-03-31:
Hi. My college is using two types of wifi network that requires log-in ID and password or log-in through webpage.
But basically octopi-network.txt file do not include script to add 'id/password', and it has only SSID name.
Is there any simple way to add id and password for wifi network?
Actually I am really new at Linux system.
Thanks.

KairuByte, 2017-04-01:
I'm not a dev of OctoPi but this type of login is not always easy to program for. Since login pages will vary depending on the underlying technology, and can even be created by the network admin themselves, there isn't going to be a one size fits all solution.

There are currently solutions out there for this, however they will likely require a knowledge of scripting, and you will need to dissect the login page to write the script. One such answer can be seen here: http://unix.stackexchange.com/a/118315

John-Mc, 2017-04-05:
This would not work on a headless pi, but is there a way to turn things over to the user for a "manual login" in situations such as this?

KairuByte, 2017-04-06:
You can, of course, run the included install script for desktop and log in as normal using the browser of your choice, but you would still need to log in after every expiry (Most AP's have a time limit before forced renewal) or script it.

guysoft, 2017-11-01:
Can I close this? Closing if no reply