guysoft/OctoPi #335: Option to disable automatic "expand File System"?

John-Mc, 2017-04-01:
I'd love to see an option to disable the automatic expansion of the file system to use the whole SD card on first launch of OctoPi. (Or failing that, a SIMPLE method to "de-expand" it.)

I volunteer at a local school and help set up their 3D printer. I typically work with 32 GB SD cards, so the file system gets expanded to close to 32GB when I launch OctoPi and enter all of the setup information, install the plugins the school uses, etc.  I like to keep a backup image, so we can quickly restore things to that base setup if a card gets damaged or lost. Unfortunately, the backup images are also 32MB. I know I can compress them for storage, but I still need a 32MB card to install them on if I want to set up a new system. In addition, since all SD cards are not created equal, even some 32GB cards will not accommodate this 32GB image.

If initial image included a bit of free space so I could do set up and install a couple of plug-ins, and I could (temporarily) disable the auto-expansion, I could get set up and make an image that would fit on smaller cards if needed.

Failing that, a simple method for shrinking the image after I'm all set up would work. I've found a few descriptions on how to do this, but they are not tailored to use on a Mac (and I've been told doing it on a Pi is painfully slow). It's also a complex procedure. Not something I can do from memory. 

So is there a simple way to disable the automatic expansion (hopefully one that I could do from memory), of if someone has a link to some very user-friendly methods of shrinking an image (preferrable a program to run, rather than a serie of commands to enter), I would greatly appreiate it.

I realize the "shrinking an image" option is not really an OctoPi-specific thing

ChinookTx, 2017-04-01:
Can't you edit the /boot/cmdline.txt and remove the resize command from there?  I have not tried it but a few posts online seem to suggest this.

John-Mc, 2017-04-01:
Thanks for the suggestion, I had no idea where the command that does this is located.

Unfortunately, I also have no idea what to edit to stop the file system expansion.
Here's the contents of my /boot/cmdline.txt 
`dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
`

John-Mc, 2017-04-01:
Sorry. Stupid question. Found the related commands in /boot/octopi.txt ... Right at the top of the file, nicely commented to tell you how to do it.

John-Mc, 2017-04-01:
re-opening:  Oops! it is in the the /boot/cmdline.txt file. I had forgotten I was on an SD where OctoPrint was installed into the regular Raspian-Jessie image.

It looks as though 0.14rc1 does have it in the cmdline.txt file. That file is identical to what was posted above, with the exception of the addition at the end of `quiet init=/usr/lib/raspi-config/init_resize.sh`.  Is it possible to comment this out somehow, so I can temporarily disable it, but still have it there?  What about if I just introduced an intentional misspelling in "init_resize.sh" to temporarily disable it, then corrected it when I wanted it to run normally?  (Not being at all familiar with what this does, I hesitate to make changes that may cause things to lock up.)

I would like to see a method of turning auto-resizing on and off in a way that was a bit more understandable to those not so fluent in Linux.

John-Mc, 2017-04-05:
Well, turning off auto-file system expansion turned out to be easier than I thought. Unfortunately, it doesn't solve my problem. Even with and unexpanded file system, if its on a 32GB card, I'll have a 32GB image when I make an image to back up. I can't install that on a smaller card.

I realize that this is not directly an OctoPi problem, so I'll consider this issue closed for th.e purposes of OctoPi

closed 2017-04-10 by John-Mc