guysoft/OctoPi #338: Rebuild

chevellebro1, 2017-04-05:
Is there a way to rebuild the installation after making changes to an existing octopi installation on a raspberry pi?

John-Mc, 2017-04-05:
If so, would doing it this way avoid the need to clear out the "known hosts" file on client computers when I make a new setup for OctoPi? It's not a big deal at home, but if I set up a new card with changes at school, I have to fix "known hosts" on 20+ laptops at the school where I volunteer (since I have no way of knowing which ones have already been used to SSH into the Pi).

Maybe this is more a Raspian thing or a client computer thing than an OctoPi thing? (For understandable reasons, the IT guy won't let us turn off "strict checking"

guysoft, 2017-04-06:
No, OctoPi uses scripts to open a raspbian image, edit it, and then close it. In order to modify it you can create a variant and use that. See [building octopi variants in the readme](https://github.com/guysoft/OctoPi#building-octopi-variants).
Alternatively, you could use Disk Dump dd (or WinDD for windows), in order to create a clone of your system.

closed 2017-04-06 by guysoft