guysoft/OctoPi #339: IP address info on Boot screen?

John-Mc, 2017-04-08:
I've noticed that the login screen now shows ways of accessing the Pi (octopi.local and the current IP address). However, the boot screen only shows octopi.local. Would it be possible to add the IP address to the boot screen as well, or does the Pi not know its IP address at the point where this text is displayed?

I'm helping out with a classroom full of kids, and an IT guy who will not give us a static IP address. Our present method is setting up an automatic email upon boot-up to a class-wide email address where the kids can check in, but that did take a bit of setup to arrange.

TheCatMan2000, 2017-04-09:
There is a plugin that you can add to OctoPrint that will show your Pi's IP Address on your printer's LCD. I hope that helps...

John-Mc, 2017-04-09:
thanks. I missed that one

TheCatMan2000, 2017-04-10:
There's also a plugin that displays M117 commands (sends text to your printer's LCD) in the web interface.  You could use that in conjunction with the other, and find out the IP through octoprint.local.

John-Mc, 2017-04-10:
I'm guessing the utility of what I'm asking is minimal. If you don't have Bonjour or some other method set up to allow accessing OctoPrint via octopi.local, you need to know the IP address. If you can see the boot screen (which probably means you are not operating headless), you can learn to type in `hostname -I` to find the IP address.

guysoft, 2017-11-01:
Fixed in https://github.com/guysoft/OctoPi/commit/4dcd1839c859098639b7c8cf74445458e887d61a

closed 2017-11-01 by guysoft