guysoft/OctoPi #340: Prints Randomly Fail

TheCatMan2000, 2017-04-09:
Occasionally when I print, Octoprint fails out on a print with the state changing to: "Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1792".

I wasn't sure where to go for this issue, but the guidelines for the Octoprint repository said "This is not the repository for Octopi," so I came here. Hope this log will help.
[octoprint.txt](https://github.com/guysoft/OctoPi/files/907969/octoprint.txt)

I'm still experimenting with different configuration cocktails to figure out what works and what doesn't. I have a Monoprice Maker Select v2 (Wanhao Duplicator i3). Right now, I'm trying to use a cable with a ferrite bead near the pi, stepper timeout disabled, and no SD card in the printer.

guysoft, 2017-04-10:
Its not an OctoPi issue, unless the serial driver or something like that.
But more probable its a hardware issue with the printer or power to the Pi. Have you tried any other host controller?

guysoft, 2017-04-10:
See this: https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception

closed 2017-04-10 by guysoft