guysoft/OctoPi #341: Weird problem running shutdown script

muelli01, 2017-04-11:
Hi all, please forgive me for posting this weird problem, but it drives me nuts.
I want to execute a script on system shutdown and neither of the known locations work on OctoPi.

When I place it in /etc/init.d and symlink it to /ect/rc0.d it is not executed (does not matter if K01 or anything in between to K99).
When I try to have systemd run it and place it in /lib/systemd/system-shutdown/ as mentioned in the systemd manual:

    Immediately before executing the actual system halt/poweroff/reboot/kexec systemd-shutdown will run all executables in /usr/lib/systemd/system-shutdown/ and pass one arguments to them: either "halt", "poweroff", "reboot" or "kexec", depending on the chosen action.

it still does not run.
When I execute the script "by hand" it runs fine.

Is there something different in OctoPi that I am missing?

Thanks for advice.

guysoft, 2017-04-11:
No, OctoPi is basically raspbian lite with a few minor changes, you can see the code in this repo

muelli01, 2017-04-11:
Thanks, but then I am kinda lost here...

guysoft, 2017-04-11:
Does it work on the raspbian lite image?