guysoft/OctoPi #343: Cannot complete load of Slic3r and Slicer Plugin on Pi Zero with WiFi

impartit, 2017-04-13:
Trying to install Slic3r using instructions at "https://github.com/OctoPrint/OctoPrint-Slic3r"  
After 12 hours or more the script errors with 
cc -I/usr/lib/arm-linux-gnueabihf/perl/5.20/CORE -DVERSION="0.01" -DXS_VERSION="0.01" -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o buildtmp/XS.o buildtmp/XS.c
The XS/C++ code failed to compile, aborting
error building buildtmp/XS.o from 'buildtmp/XS.c' at /usr/share/perl/5.20/ExtUtils/CBuilder/Base.pm line 175.

This after a long protracted attempt at installing Slic3r with the Slicer plugin on OctoPrint. Pi OS is the latest OctoPrint version 1.3.2, 0.14.

Please, please advise

fushini, 2017-05-04:
I see the same error on a pi zero w installing alic3r. Following the instructions at:
https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi

ChinookTx, 2017-05-05:
Do you run out of memory?  Did you check your SD card to see if it runs out of space?

eyal0, 2017-12-01:
@fushini I am the maintainer of those build instructions.  I was able to build prusa3d slic3r just a few days ago.  If you can try again and provide a log of the build process, that would help me out and I could make the instructions better!

guysoft, 2018-02-07:
I am closing this and moving conversation to #444 is that is ok with evreyone