guysoft/OctoPi #344: Octopi enterprise wifi

ozkalt, 2017-04-13:
Hi everyone!
I am using Raspberry Pi 3 with Octopi. I have connected wifi but I need to connect to enterprise wifi. I have changed wpa_supplicant.conf and interfaces files. It does not help to connect wifi. Does anybody have this situation?

guysoft, 2017-04-13:
What does enterprise WiFi mean?

bardiir, 2017-04-13:
Probably something with a certificate authentication like 802.1X - but I'm unsure if this really is an issue with OctoPi, this is probably more likely something that would need to be solved in the underlying linux distribution.

ozkalt, 2017-04-13:
Enterprise WiFi means enterprise network. I mean, I have tried to connect using network of Campus. That network is like "eduroam".

ozkalt, 2017-04-13:
I solved the problem using another methods. You can use this way: https://gist.github.com/chatchavan/3c58511e3d48f478b0c2

closed 2017-04-15 by guysoft