guysoft/OctoPi #348: wifi channel 12/13 don't work (set wifi region from octopi.txt?)

schwabe, 2017-04-18:
I am running octopi-jessie-lite-0.13.0.zip and wifi channels 12/13 did not work and were not even listed in ```iwlist channel```. Found that this is a common issue with rpi. I set the wifi region with raspi-config and updated the kernel/distribution with sudo apt-get dist-upgrade. Channel are now listed in ```iwlist channel``` but it still does not connect to my wifi on boot if it is on channel 12 or 13. I workarounded this by changing my wifi channel.

I think the wifi region should be settable via a txt file in boot or default to something that also allows channel 12/13.

John-Mc, 2017-04-18:
I think this can be set by editing the /boot/octopi-wpa-supplicant.txt file. Somewhere in the first line or two it sets the country code for WiFi operations. The default is GB.

NOTE: I believe the /boot/octopi-wpa-supplicant.txt file **does not exist** in OctoPi version 0.13. I'm seeing it in 0.14rc1, so maybe easier access to this setting is coming soon?

guysoft, 2017-04-19:
This issue is fixed in 0.14.0

closed 2017-04-19 by guysoft