guysoft/OctoPi #349: Cant connect to Renkforce RF100 printer

cr0co, 2017-04-23:
Hello

i successfully installed Octopi on Raspberry Pi 3
interface and webcam works fine.

But i cant connect to my Renkforce RF100 printer

Serialport: Auto (i changed it to /dev/ttyUSB0 with no success)
Baudrate: Auto

in terminal:
`Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x76225630, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 250000
Send: N0 M110 N0*125
Recv: \x1e\x7f�\x7f\x06|\x18\x7f�\x7f\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x1es\x00\x06�\x06���\x06\x1e\x00\x18�\x18\x7f\x18�\x18\x7f\x18�\x0c\x00\x18\x0c��\x06��\x06�\x06��\x06��\x1e\x18\x18\x00\x18|�\x03>\x1e\x7f�\x7f\x00\x18�s\x00\x06���\x06���\x06�\x06�\x0c\x00\x18\x7f��\x1e�\x7f\x00\x18�\x18�\x18\x00\x18�\x18�\x18�\x18�\x18\x00\x18�\x18�\x0c���\x0c\x1e��\x0c�\x0c�\x18\x00\x18�\x06�\x18\x009�\x06�\x06��\x1e\x18\x7f\x18\x18\x00\x18p\x18\x7f�\x1e�\x1e�|�\x06\x00\x18\x0c��\x06�s���\x06�\x06�\x06�\x18��\x1e�\x1e�|\x06\x1e�|\x06\x06\x18\x00\x1ep�\x1e�\x1e\x00?\x06\x1e�\x1e�|�|\x18\x18\x00\x18\x7f��\x1e�\x7f\x00\x18�\x18�\x18\x00\x18�\x18�\x18�\x18�\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x0f�\x1c����\x06�\x18\x1es\x06���\x1e\x18\x7f\x06\x06�\x0c\x00��\x0f�\x0f\x1e\x0f�\x0f\x00\x18\x00\x18���\x1e\x06|�\x1e�\x1e�|\x18\x7f\x18pf?�|�|�|\x18\x7f\x18p\x06\x06�����\x06�\x0c\x00�\x1e\x0f\x00\x07\x00\x18s��|\x1e|\x00\x1e�\x1e\x18\x18p�\x03>\x18\x7f�|\x1e|�\x1e�|�|�|\x00\x18\x0c�c>�|\x06|�\x7f�\x1e�\x7f\x00\x18\x18��\x06�\x06���\x06\x1e�|\x1e\x7f\x00\x18|��\x1e\x06|�|�|�|\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x18\x06���\x06���\x06�\x18���\x06�\x06�\x18�\x06\x1e\x06\x1e�\x7f\x18\x18s�g<\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es�\x18�\x18\x00\x18���\x0f\x00\x18\x7f\x18�\x18�\x18\x00\x18���\x0f\x00\x18\x7f\x18�\x18�\x18\x00\x18��\x18\x0f\x00\x07\x00\x07�\x06\x00\x07\x00\x07\x00\x18\x06�\x06\x0f\x00\x07�\x0f�\x06\x00\x07\x00\x07\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x1es\x00\x06���������\x06�\x18\x0f����\x06�\x06���\x06�����\x06�\x18p\x18\x1e���\x06\x1e\x7f\x06\x06\x18\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es\x06\x18�\x18�\x0c\x00\x18��\x1e\x0f\x00\x07\x00\x07�\x06\x00\x07\x00\x07\x00\x18��\x1e\x0f\x00\x07\x00\x07�\x06\x00\x07\x00\x07\x00\x18��s\x18\x7f\x18�\x18�\x18\x00\x18\x06�\x18\x0f�\x0f�\x06\x00\x07\x00\x07\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x1es\x00\x06���������\x06�\x18\x009�����\x06���\x06���\x06����\x1e�\x1e\x00\x18p\x18\x1e���\x06\x1e\x7f\x18\x0f\x06\x06\x18\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es\x06\x18�\x18�\x18\x00\x18���\x0f\x00\x07\x00\x07\x00\x18���\x0f\x00\x07\x00\x07\x00\x18��\x06\x0f\x00\x07\x00\x07\x00\x18\x06�\x06\x0f\x00\x07\x00\x07\x00\x07\x00\x07\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x009�����\x06���\x06���\x06����\x1e�\x1e\x1e\x7f\x18\x18\x00\x18���\x18�\x06���\x06\x1e�\x7f\x06\x1e�\x1e�|�\x06\x00\x18���\x18���\x06�\x06�����\x06�\x06�\x18\x00\x06\x1e�|\x00\x18���\x18�\x06���\x06�s�\x03\x1e\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es\x06\x18�\x18�\x18\x00\x18���\x0f\x00\x07\x00\x07�\x06\x00\x07\x00\x07\x00\x18��\x1e\x0f\x00\x07\x00\x07\x00\x07�\x06\x00\x07\x00\x07\x00\x18���\x0f\x00\x07\x00\x07�\x06\x00\x07\x00\x07\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x009\x1c\x06�s�\x06\x1e\x1e|�|�|\x00\x18���\x06�����\x06�\x06�����\x06�\x0c\x00\x18\x18��\x18\x1es\x18\x06\x1e\x00\x18\x0f����\x06�\x06���\x06���\x06�\x18p\x18\x1e���\x06\x1e\x7f\x06\x06�\x06\x00\x18���\x18\x1es\x18\x06\x1e\x00\x18�\x06���\x06�s�\x03\x1e\x00\x18\x0f����\x06�\x06���\x06���\x06�\x18p\x18\x1e���\x06\x1e\x7f\x06\x06�\x06\x00\x18\x03��\x18\x1e��\x06\x1e\x06\x1e�\x1e�\x7f�\x1e\x00\x18�����s\x1e�|�\x1e�\x7f\x00\x18�������\x06�\x18p\x18\x1e����\x18|\x18\x00\x18���\x18\x1e��\x06���������\x06�\x18����\x00\x18s��\x06���\x06�\x18p\x18\x1e���\x06\x1e\x7f\x06\x06�\x06\x00\x18\x00\x18���\x18\x1e��\x06���������\x06�\x18��\x00\x18s��\x06���\x06�\x18p\x18\x1e���\x06\x1e\x7f\x06\x06�\x06\x00\x18\x00\x18\x06s�\x18\x1e��\x06���������\x06�\x18\x06s\x00��\x1e�|\x18\x7f\x1e\x1e\x00\x18p\x18\x1e���\x06\x1e\x7f\x06\x06\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es\x06\x18�\x18�\x18\x00\x18\x18\x06�\x18\x7f\x18�\x18�\x18\x00\x18��\x00\x18\x7f\x18�\x18�\x18\x00\x18\x03�\x18\x0f\x00\x07\x00\x07\x00\x07\x00\x07\x00\x18��\x18\x0f\x00\x07�\x06\x00\x07\x00\x07\x00\x18��\x00\x18\x7f\x18�\x18�\x18\x00\x18\x06s�\x18\x7f\x18�\x18�\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18p��\x1e�\x1e�|\x00\x18\x1e�|�|\x1e\x7f�|�\x7f\x00\x18p\x18\x1e����\x18�\x18s�g<\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es\x06\x18�\x18�\x18\x00\x18��\x00\x18\x7f\x18�\x18�\x18\x00\x18�s\x00\x18\x7f\x18�\x18�\x18\x00\x18��\x00\x18\x7f\x18�\x18�\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18\x1es\x00\x06���\x06�����\x06�\x06�\x18p����\x06���\x06�\x06�\x18���\x06�������\x06���\x06���\x06�\x0c\x06�\x06��\x06��\x1e\x18\x18\x00\x18\x00\x18\x1es�\x18�\x18�\x18\x00\x18\x1es\x00\x18\x00\x18p�\x18\x0f\x18\x0f\x00\x07\x00\x18\x03�\x7f\x18�\x18\x00\x18\x0f�\x00\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es�\x18���\x18\x00\x18\x1es�\x18\x00\x18p�\x18\x0f�\x0f\x00\x07\x00\x18\x03�\x06\x0f\x06\x0f\x00\x07\x00\x18\x0f�\x00\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18���\x18�p\x00\x18���\x06�\x06���\x06\x1e�|\x1e\x7f\x18\x18s�c<\x1e|\x00\x1e�\x1e\x18\x18\x00\x18\x00\x18\x1es�\x0c�\x18�\x18\x00\x18��\x18\x0f\x18\x0f�\x06\x18\x0f\x00\x07\x00\x18\x18s�\x18\x7f\x18�\x18�\x18\x00\x18\x0c�\x06\x0f\x06\x0f�\x0f�\x06\x00\x07\x00\x07\x00\x18\x00��\x18\x7f\x18�\x18�\x18s�c>\x1e|\x00\x1e�\x1e\x18\x18\x0f��\x1e�\x1e\x06|�\x1e�|�\x1e�\x7f\x00\x18���\x06�\x06���\x06\x1e�|\x1e\x7f\x18\x18\x00\x18\x0c��\x1e\x1e\x7f\x06|\x18|�\x1e�|�|\x18\x00�|\x1e|\x00\x1e�\x1e\x18\x18\x18\x06\x03\x06�\x18���\x06�\x06�\x06�\x18\x1e\x1e\x1e\x18\x00
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 230400
Send: N0 M110 N0*125
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 115200
Send: N0 M110 N0*125
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1792
Changing monitoring state from 'Detecting baudrate' to 'Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1792'
Connection closed, closing down monitor
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6bac75b0, open=True>(port='/dev/ttyUSB0', baudrate=57600, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: ��\x0f�s��c\x0d�r!mp\x1a���\x1e\x00��M��X
Recv: Jj\JkO
Recv: \x0e���\x0b��,�������G�\x19Y[\x14�\x1e�I\x08�wx\x14Bzpx�N�kj\x08�����o\x09���I
Recv: ���������������\x0f��� 2�������L��M\x08\\x09�S���*J�K,J�L+H��\x1d\x13x\x7fP{���\x08�\x14���\x1d�\x17\x0b�I*J�J*H�(J�zI
Recv: ������\x02(��!�\x11��6�wx\x16��\x08�J\x0c��B~�B�JL
Recv: ���������
Recv: ~��b�\x09\x7f���\x0f�\x0c\x7f����0\x1d�\x17\x09�H*J�MJK
Recv: ��rJ��M����R��0uIt\Pz��\x0c�<\x1d\x0e\x7f�\x0f�������\x0d�<\x1f\x08}���\x0c���\x0e��\x08�*������\x0e��\x08��(\x08~���\x0c\x0f��\x08�\x14\x18��\x11y\x7fp\x10�\x18�z�\x05\x1a�\x08D\x08HT\x08\x09T\x08I�JJ
Recv: �!\x10�T\x08\x09W\x08\x06�O�����\x0f�
Recv: �s
Recv: �I\x09NI
Recv: NI
Recv: NI
Recv: ������C\�s!&���9 0\x14�j
Recv: L�I
Recv: |
Recv: M��M\x08]{�
Recv: �!��\x08H\x14Hx�X��>z�\x06�M\x08]Z�KI
Recv: \I
Recv: �Z*H�(J��M��������O����\x0d���\x0c #��
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor`

dtynan, 2017-04-24:
It looks like your baud rate is wrong. Try manually setting it to 38400. At the very high baud rate (250k) in the first example, you get a lot of data back suggesting that it's too high. The 57.6k speed doesn't produce as much line noise. Try 38400, and work down from there in factors of 2 (19200 9600 etc).

cr0co, 2017-04-24:
Hello dtynan

I tried, but it failed :(

`Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6b7df6b0, open=True>(port='/dev/ttyUSB0', baudrate=38400, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1d=�):��\x18\x08�1���H���\x0fO|��r2y;8�;=��2����OJq:(��)�:;�:���X���r�������#�9u2�xKQ�\x05�r`r����\x0b�\x1e)�8��\x19���0��X`�X`�`=b��a`\x1d���eB9`��`�0_\x0b�\x0b\x01�a\x1d��\x19�9�\x00tr2;����1\x07��`�X\x07���^!r\x1d�\x19�\x19\x1d\x07���eP�����Y�e@��0����\x19k��029e�g� �0e`g���0e^1\x07@bPb��0\x1f���0b��;���wB9��������y�A��#��2��Q1\x05P���p 9`+��\�X\x07Q!���!9`X�8\x18J��\x0b`��\x1e)3�\x18r�\x149��!p\x0b��
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6bd250b0, open=True>(port='/dev/ttyUSB0', baudrate=19200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: �\x14i%�o\x0eb:�)� �2bh8\x1e9\x13�� <�}��(h�4�!bg2�=\x1e�`\x1f�bs2�-J}�\x14������J�8l��\x14�cr\x1e\x03\x16J�6=���8��?�,7/���#��� ��a!! "!3���\x14Psr�~���\x14�xZ��b\x08���\x14�,y\x1e��O�,�4V�
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6bd25390, open=True>(port='/dev/ttyUSB0', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
[... cut ...]
Connected to: Serial<id=0x6b814ed0, open=True>(port='/dev/ttyUSB0', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: $�\x13���1a���\x04�\x17Q�\x0b8��p�\x13��\x06\x16p1�\x13x���d�d0����\x12�p\x12q0\x089���81
There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor`

dtynan, 2017-04-24:
Have you connected to the printer directly? If so, can you pull up the baud
rate you used?

Failing that, continue down the scale with lower baud rates. 4800, 2400,
1200 and even 300. See if they work.

It does look like a baud rate issue, but it's also possible you have a
noisy serial line. I'm not familiar with your printer type. Are you sure it
has an ASCII interface? I had a quick look at the online manual, and it
suggests a baud rate of 115k, which didn't work for you, which would
suggest either some sort of binary interface or a noisy line.
        - Der

cr0co, 2017-04-24:
yes printer is diretly connected to the pi

in my octoprint, 9600 is the lowest ?

i dont know its interface :( i hopd it works..

John-Mc, 2017-04-24:
I think @dtynan was asking if you have connected your printer directly to your **computer**, not to the Pi.  Whether or not is works correctly when connected to your computer will help narrow down the problem.

BTW, are you using a good-quality USB cable?

cr0co, 2017-04-24:
Hello

Yes i printed diffrent things directly from the computer or SD card.

i use the usb cable that came with the printer, should be good

guysoft, 2017-04-25:
This is not an OctoPi issue, its an OctoPrint issue. Please post it in the OctoPrint issue tracker:
https://github.com/foosel/OctoPrint/issues/

closed 2018-01-07 by guysoft