guysoft/OctoPi #350: Slicing failed

gyrex, 2017-04-24:
Hi guys,

I'm having some issues with octopi slicing. It used to slice ok but now it won't slice anything, including previously sliced objects. For this particular print I was trying to slice 3DBenchy.The error I'm getting via the GUI is:

Slicing failed
Could not slice Spool_Holder_Left.STL to Spool_Holder_Left.gco: Unknown error, please consult the log file

Log file:

2017-04-25 02:50:12,690 - octoprint.plugins.cura - ERROR - Could not slice via Cura, got an unknown error
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/cura/__init__.py", line 296, in do_slice
    self._logger.info(u"Running %r in %s" % (" ".join(args), working_dir))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 334: ordinal not in range(128)

Any ideas? In the meantime, I'll slice in Cura on my desktop.

guysoft, 2017-04-25:
This is not an OctoPi issue, its an OctoPrint issue. Please post it in the OctoPrint issue tracker:
https://github.com/foosel/OctoPrint/issues/

closed 2017-04-25 by guysoft

gyrex, 2017-04-25:
Thanks guysoft. I was going to post it there but they explicitly say to post all issues with OctoPi installs here. Anyway, I'll post it there.

guysoft, 2017-04-25:
Who is they? At what stage?

gyrex, 2017-04-25:
"They" being the moderators of the OctoPrint github.

From: https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md

"This is not the Github respository of OctoPi, which is the preconfigured Raspberry Pi image including OctoPrint among other things - that one can be found here. Please note that while we do have some entries regarding OctoPi in the FAQ, any bugs should be reported in the proper bug tracker which - again - is not here."

guysoft, 2017-04-25:
Ok, and your issue is not an OctoPi one
cc @foosel

gyrex, 2017-04-25:
Yes, I understand this and agree with you but the terms stated above seemed pretty clear about posting any issues here. It didn't make sense to me. In any case, I've filed the bug on the OctaPrint github. Thanks for your help.