guysoft/OctoPi #351: Is there a guide for install cura or repetier on octoPi system?

Jolly2015, 2017-05-02:
Hi there, last day I downloaded octoPi .img to my raspberry 2b,and I noticed curaEngine 15.04 can slicing directly on this system like “http://octoprint.org/download/” showed,so I would like to know is there a guide for how to install slicing sofeware on raspberry.Thanks.

John-Mc, 2017-05-02:
@Jolly2015   The Cura 15.04 engine is packaged with the OctoPi image, so it's already installed. You can slice on your regular computer and send the g-code file over to the Pi (bypassing the need to slice in OctoPrint), or you can send an STL file to the Pi/OctoPrint and slice it there.

To slice in Cura on the Pi/OctoPrint, you'll first need a version of Cura on your desktop/laptop computer. You want version 15.04.xx or **older** to set up your slicing profiles (newer versions of Cura are not compatible with the version currently bundled with OctoPrint). You can find Cura for Windows, Mac or Linux OS [here](https://ultimaker.com/en/products/cura-software/list). Install the appropriate version on your computer, run it and make all the settings you wish (nozzle temp, bed temp, layer height, etc), choose "save profile" from the file menu and give it a name. This will create a ".ini" file. This is the file you want to import into Cura. I usually make several profiles for each filament I will be using (for example: PLA standard, PLA fast, PLA detailed). 

OctoPrint has no provision for modifying profiles, so you would need a separate profile for each adjustment. So if I wanted to turn supports on, I'd need to add a "PLA Standard supports", "PLA fast supports". If you do much slicing within OctoPrint, this can get cumbersome. Fortunately, the "Full Featured Slicer Plugin" will add another tab to OctoPrint which allows you to adjust a number of parameters. I could probably get by with just importing 1 profile into OctoPrint and making changes to that for 90+ percent of my printing (however, I still prefer to have a few profiles set up for each filament type. I find it much easier to start form a known good profile and just tweak it)

Lastly: Now that I've hijacked Guysoft's github page enough... I'd suggest looking at [OctoPrint.org](http://octoprint.org/) for documentation and FAQs, or at the  [OctoPrint Google+ Community](https://plus.google.com/u/0/communities/102771308349328485741), or the [OctoPrint User's Group](https://www.facebook.com/groups/piprint/) on Facebook for more in-depth help. I may be wrong, but I believe this github page is more intended for bug reports, feature suggestions, and other development related work.

guysoft, 2017-05-10:
@John-Mc Its contribution not hijacking :+1: 
I personally have no time to write all answers, and the ones you are posting and yours are excellent.
Also closing.

closed 2017-05-10 by guysoft