guysoft/OctoPi #352: Request: How to make a script run at startup

AbuMaia, 2017-05-03:
Pretty much what the title says. How do I make a BASH script run at startup? I'm using WiringPi to control the GPIO pins to control a relay board. I need to ensure that the GPIO pins are in a certain configuration upon startup in order for the relay board to work properly.

John-Mc, 2017-05-03:
I'm guessing this is not really OctoPi related? I know zip about BASH scripting, but here's a description for sending an email with the pi's IP address on startup that might help you figure it out. You can skip the video (most of which is about the email script itself) and go right to the links shown in his text description below. He lists two methods of running something at boot in his written documentation. I used the "older method" which calls the script from within /etc/rc.local  It's been working fine for me:  [Email Pi's IP at startup](https://www.youtube.com/watch?v=g940HdYWBTo&feature=youtu.be)

AbuMaia, 2017-05-05:
It's OctoPi related in that OctoPi seems to be the OS of the system, or had modified the OS, and I was asking for assistance in how to use the OS to get this to work through OctoPi. A general search for this on Google, which gave results regarding Raspbian, mentioned folders that weren't present in OctoPi.

John-Mc, 2017-05-05:
Raspian is the OS used for OctoPi, but it's a stripped down version, so not averything from the "standard" Raspian is available.  The link  posted above works on my OctoPi installation

AbuMaia, 2017-05-05:
Really stripped down. I don't seem to have an /etc/rc.local either.

closed 2017-05-05 by AbuMaia