guysoft/OctoPi #353: octopi.local disappears

tcurdt, 2017-05-05:
I just installed octopi "Version 1.3.2 (master branch)". When the pi starts up I can see `avahi-daemon` announcing the octopi http service.

```
OctoPrint instance on octopi
```

Unfortunately after a few minutes the http service disappears.
The workgroup manager service is still listed though.

And `avahi-daemon` is still running
```
pi@octopi:~ $ ps ax | grep av
  355 ?        Ss     0:00 avahi-daemon: running [octopi.local]
  399 ?        S      0:00 avahi-daemon: chroot helper
```

Restarting the service

```
pi@octopi:~ $ sudo service avahi-daemon restart
pi@octopi:~ $
```

unfortunately it does not help to announce the http service again.

guysoft, 2017-05-05:
You posted the version of OctoPrint you are using - what version of OctoPi
are you using?

tcurdt, 2017-05-05:
Ah, sorry about that. This is with a fresh install of OctoPi 0.13.

tcurdt, 2017-05-05:
This could possibly be applicable:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=736641
https://bitbucket.org/marc_culler/querierd/
https://github.com/machinekoder/querierd

slider162, 2017-05-06:
Is it avahi-daemon or haproxy that won't start?  What happens if you restart the haproxy service?  Mine is failing to start after an update/reboot.

tcurdt, 2017-05-06:
I've restarted the whole network including the router and it *seems* to be OK now. Very odd. I'll watch this closely and re-open if I see it again. Sorry for the noise if it was just a glitch on the router.

closed 2017-05-06 by tcurdt