guysoft/OctoPi #354: Octoprint not running

Penfold81, 2017-05-08:
Hey, noob here with a problem with octoprint on a pi 3. its been running fine for months but yesterday my pi updated and now octoprint wont startup? the logs show me a blank screen over SSH and when i run "ps -ef | grep -i octoprint" octoprint is in red.

im now lost

i dont really want to wipe the card and start again as im running my lights through it too and have setup buttons to activate gpio pin 18 and cant really remember how i done it

HELP!!

ChinookTx, 2017-05-08:
If you see it in red it's because you grepped for it and grep highlights what it found.  The fact it shows up means it is somewhat running.  Maybe posting the octoprint.log (get it from the .octoprint/logs folder) would help pinpoint what goes wrong, if anything.

Penfold81, 2017-05-08:
how do i get into the logs? it just shows a blank screen on SSH?

John-Mc, 2017-05-08:
A tip for the future (which won't really help you now):

When I'm modifying things on an SD card - as you did with your light controls and setup buttons, I create a ReadMe text file and save it in the Boot directory. Since the Boot directory is accessible when the card is plugged in to my laptop, I can see it even if I can't get my Pi running. (I include my initials in the filename, so I know it's one that I made: ReadMe_JPM.txt)  I put in what image I started with and everything I did to set that card up, including if I changed the Pi's password from the default, any OctoPrint passwords, plugins added, or any other modifications to the stock image. I also include the URLs to any websites or YouTube videos I used to aid in the customization. It makes it easy to recreate if I have to.

And of course, once you have everything set, **make a backup .img file**!  If you are using a Mac, Apple Pi-Baker works well for this. I'm sure there are similar programs for Windows or other systems.

Penfold81, 2017-05-08:
good to know cheers, wish i knew i could back it up. its a mac yeah

Penfold81, 2017-05-09:
i have tried for hours with this now. cant seem to get it to startup again?

Penfold81, 2017-05-10:
anyone?

guysoft, 2017-05-10:
What is the output of `df`?

Penfold81, 2017-05-10:
i dont understand?

Penfold81, 2017-05-10:
pi@octopi:~ $ df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       15374768 14852580         0 100% /
devtmpfs          435672        0    435672   0% /dev
tmpfs             440280        0    440280   0% /dev/shm
tmpfs             440280     5972    434308   2% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs             440280        0    440280   0% /sys/fs/cgroup
/dev/mmcblk0p1     61384    22320     39064  37% /boot
tmpfs              88056        0     88056   0% /run/user/1000

ChinookTx, 2017-05-10:
Your root ( / ) partition is full. You need to free up some space.

Penfold81, 2017-05-10:
how do i do that? is there a file on the root i can access if i plug it into my mac? or can i do it over ssh?

Penfold81, 2017-05-10:
ok i have sorted it i think. i ran 
sudo mount --bind / /mnt
sudo ncdu -x /mnt
then went through all temp files and time laps files and deleted them

peterlpop, 2019-03-18:
and you did that how?