guysoft/OctoPi #355: how is octoprint running in octopi

pokemonspeler, 2017-05-09:
i was searching for a possibility to add a custom webpage to my raspberry pi 2b but i cant find where octoprint is running. of course i can install apache but in this case i would need to use another port witch is not so handy. please does someone now where to find this, or how to fix that apache can be used from port 80

guysoft, 2017-05-10:
OctoPrint is running using flask and HAproxy. You could install your
favorite webserver (and make it listen to a custom port, such as 8000), and
then modify HAproxy settings to redirect to it from a certain url path.

Guy

pokemonspeler, 2017-05-10:
is it posible to put a link here or a tutorial on how to do this?

guysoft, 2017-05-15:
See https://github.com/guysoft/OctoPi/blob/devel/src/filesystem/root/etc/haproxy/haproxy.cfg
Look at the ``backend webcam`` section, you can forward it to a path and port.

For example, port 8000:

backend webcam
        reqrep ^([^\ :]*)\ /insert_name_here_to_add_to_url/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8000

guysoft, 2017-05-22:
Closing unless someone objects

closed 2017-05-22 by guysoft