guysoft/OctoPi #356: C270 Webcam

abreumateus, 2017-05-19:
##What were you doing?

Installing camera on Octoprint.

##What did you expect to happen and what happened instead?

The webcam just worked

The webcam didn't worked

##Branch & Commit or Version of OctoPrint

1.3.2

##Operating System running OctoPrint

Octopi on a Pi Zero

Power supply 5V 1.0A

##Printer model & used firmware incl. version

Marlin

##Browser and Version of Browser, Operating System running Browser

Google Chrome

##Just did this tutorial

Webcam
If you also want webcam and timelapse support, you'll need to download and compile MJPG-Streamer:
cd ~
sudo apt-get install subversion libjpeg8-dev imagemagick libav-tools libv4l-dev cmake
git clone https://github.com/jacksonliam/mjpg-streamer.git
cd mjpg-streamer/mjpg-streamer-experimental
export LD_LIBRARY_PATH=.
make
This should hopefully run through without any compilation errors. You should then be able to start the webcam server using:
./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
Everything works untill the last comment.
Result for command "lsusb":
Bus 001 Device 005: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 004: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 003: ID 2341:0042 Arduino SA Mega 2560 R3 (CDC ACM)
Bus 001 Device 002: ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

##Result for command "./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so":

pi@octopi:~/mjpg-streamer/mjpg-streamer-experimental $ ./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
MJPG Streamer Version.: 2.0
i: Using V4L2 device.: /dev/video0
i: Desired Resolution: 640 x 480
i: Frames Per Second.: -1
i: Format............: JPEG
i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
o: www-folder-path......: disabled
o: HTTP TCP port........: 8080
o: HTTP Listen Address..: (null)
o: username:password....: disabled
o: commands.............: enabled
libv4l2: error turning on stream: Operation not permitted
Unable to start capture: Operation not permitted
i: Error grabbing frames

##Please help me! :(

ChinookTx, 2017-05-19:
If you are using OctoPi, you don't need to compile mjpg-streamer, it is
already built and configured for you.  You are likely seeing these errors
because the service is already running.  What happens when you visit
http://addressofyourpi:8080  ?

guysoft, 2017-05-19:
Indeed, OctoPi comes pre-installed with mjpg-streamer. Moreover - why did you paste the issue template from OctoPrint?

guysoft, 2017-05-19:
Dulicate of https://github.com/foosel/OctoPrint/issues/1926

closed 2017-05-19 by guysoft

abreumateus, 2017-05-19:
Because she closed my topic.

Why did you closed my topic?

abreumateus, 2017-05-19:
I'm not at home right now to answer  ChinookTx , please open my topic again dude.

ChinookTx, 2017-05-19:
I suspect your issue was closed because it is a non-issue.  mjpg-streamer is already installed in octopi (and working out of the box with a Logitech C270, this is what I'm using).

guysoft, 2017-05-19:
1. "She" is @foosel and happends to be the project leader. Who gave you explicit instructions in the issue you opened.
2. Your camera has been reported to work and this is not an issue or enhancement.
3. You are trying to install a component that already comes preinstalled in OctoPi, I have no idea what you did to your image.
4. I don't take well to being ordered to do things in my own Free Software Project, I will only listen to well framed and aargued suggestions.

guysoft, 2017-05-19:
Also what @ChinookTx said

abreumateus, 2017-05-19:
@ChinookTx 
I know, but why mine just don't work. I reinstalled the image twice and i'm having the same problem.
@guysoft 
1. But it just don't work for me.
2. Ok, but why mine just don't work? I just get out of the box, tested on my PC and then put it on my usb hub of Raspberry Pi Zero.
3. I installed a standard version of Octopi and i'm in the same condition.
4. Good for you (Y), nobody supposed to be an expert on the topic. There are a lot of dummies like me that need help, if you don't want to help, just don't. Let the others like @ChinookTx help me. Sorry to bother you, but i didn't found the solution to the problem that i'm having, that's why i created the topic, -.-'.