guysoft/OctoPi #358: Website visitors can go to /webcam/control.htm and modify webcam settings with no authentication

OmgItsBkid, 2017-05-21:
Copying this over from the OctoPrint report I put in on their tracker:

What were you doing?

Browse directly to http://yoursite/webcam/control.htm and you will be presented with webcam controls that work in real time and affect all users viewing the webcam. This can be done without the person logging into OctoPrint.

What did you expect to happen and what happened instead?

I did not expect this page to be available (as it does not appear to be linked from what I can tell), or at least available behind the OP login.

Branch & Commit or Version of OctoPrint

OctoPrint 1.3.2 (master branch)

Operating System running OctoPrint

OctoPi 0.13

Printer model & used firmware incl. version

N/A

Browser and Version of Browser, Operating System running Browser

N/A really, but Version 58.0.3029.96 (64-bit) on Windows, and Chrome 58.0.3029.83 on Android

Link to octoprint.log

https://pastebin.com/qU0kTKSj

Link to contents of terminal tab or serial.log

N/A

Link to contents of Javascript console in the browser

N/A

Screenshot(s) or video(s) showing the problem:

[image]

guysoft, 2017-05-22:
Can you please link the OctoPrint issue? If you unconnect them there is no tracking.

OmgItsBkid, 2017-05-22:
https://github.com/foosel/OctoPrint/issues/1930 This is the issue that was put in on the OctoPrint side, however they stated that it was an OctoPi issue, so it won't be looked into any further on their end.

PR #359 (2017-05-23): Custom webroot & disabled controls for webcam server
Instead of ./www we now use ./www-octopi by default which only
contains an index.html (generated during image build) that displays
only the snapshot and stream for debugging purposes.

The command action on mjpg-streamer (which control.html of the stock
webroot utilizes) has also been disabled via "-n".

Two new variables have been introduced in /boot/octopi.txt to
allow configuring the webroot to use and the additional options
for output_http.so (by default only "-n"), in case power users
need to change back or otherwise customize stock beh [... cut]
commit ef576525 (2017-05-23): Custom webroot & disabled controls for webcam server

Instead of ./www we now use ./www-octopi by default which only
contains an index.html (generated during image build) that displays
only the snapshot and stream for debugging purposes.

The command action on mjpg-streamer (which control.html of the stock
webroot utilizes) has also been disabled via "-n".

Two new variables have been introduced in /boot/octopi.txt to
allow configuring the webroot to use and the additional options
for output_http.so (by default only "-n"), in case power users
need to change back or otherwise customize stock be [... cut]
closed 2017-05-23 by guysoft