guysoft/OctoPi #361: Cannot upgrade octoprint due to power loss during upgrade

alexlinno, 2017-06-01:
This is not a bug, but didn't know where else to ask and not sure whether it is an octopi or octoprint issue:
I wanted to upgrade from octoprint 1.3.3 to 1.3.4 and was so stupid to switch it off during upgrade. When I try to upgrade now I get this message
Updating, please wait.
Python executable: '/home/pi/oprint/bin/python'
>>> Running: git diff --shortstat
> git diff --shortstat
fatal: unable to read 3a36cf58f26c4bd817850b40804344195c4c4143
Traceback (most recent call last):
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.3-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 313, in <module>
main()
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.3-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 309, in main
update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.3-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 204, in update_source
if _rescue_changes(git_executable, folder):
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.3-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 180, in _rescue_changes
raise RuntimeError("Could not update, \"git diff\" failed with returncode {}".format(returncode))
RuntimeError: Could not update, "git diff" failed with returncode 128
The update did not finish successfully. Please consult the log for details.

I guess the Git repository is corrupt. What is the best way forward? Should I just remove the /home/pi/OctoPrint/.git and clone it again???
Anyways thanks for any hint
BR
Alex Lindeijer

ntoff, 2017-06-04:
Try the manual update method as described in the wiki: https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-update-the-octoprint-installation-on-my-octopi-image

guysoft, 2017-07-23:
Any news on this? if not will close due to inactivity

alexlinno, 2017-07-31:
I just pulled the git repository and every thing was fine. Thanks

closed 2017-07-31 by alexlinno