guysoft/OctoPi #363: OctoPi Daily: Not able to SSH with RPi3

tubeme, 2017-06-05:
Hello I just downloaded today's daily build from the mirror page.

I cannot connect to OctoPi with SSH. I use Rpi3.

At first I thought it is not enabled. Made a ssh file in the boot. The file disappeared ie SSH enabled. Putty is terminating with a Fatal network error.

Before this daily I was with the 0.14 RC1 and there was no problem with the SSH connection.

guysoft, 2017-06-07:
The daily might be broken, try the next one

DarkSeverance, 2017-06-08:
I have the same issue but I didn't download from the daily. I downloaded the latest version from "http://octoprint.org/download/". The md5 has that it lists on the webpage is "d0191958de7ffe0f0a62c9313b7a3fc9", which matches the zip file that I download. When I extract it, it is labeled "2016-03-18-octopi-jessie-lite-0.13.0.img" with a md5 of "08edb5c318ae9932cc4f468ebc21e104". The image was loaded on two different SD cards, tested with two different Raspberry Pi 3. 

The first one pulled an IP address that I could see with IP Scan. When I try to connect to it, I get a connection refused immediately. I then tried to put a ssh file in the second one. This one doesn't pull an IP address at all, I can't find it on the network. Removing the ssh file shows up on the network, but still connection refused.

tubeme, 2017-06-12:
Works now.

closed 2017-06-12 by tubeme