guysoft/OctoPi #364: Disable the screen saver

Lepterois, 2017-06-11:
Hi!
Sorry I'm new with Octopi and I'm looking for disable the screen saver on the octopi 1.3.4, PI3.
I've tried many solutions, but none is working...

Do you have a solution?

Thanks

guysoft, 2017-06-11:
What screen saver?

Lepterois, 2017-06-11:
After 5 minutes without activity the screen turn off until I touch it again

guysoft, 2017-06-19:
You mean the terminal screen?

Lepterois, 2017-06-19:
I've a Pi3 with the original 7" screen where is octopi (octopi img, not rasbian + octopi) and touchui installed. This pi is connected to the 3d printer. I'm talking about this screen. Everything is working well, except after about 5 minutes the screen power turn off until I touch the screen

NF6X, 2017-07-04:
I have also been having this issue. I'm controlling my printer with OctoPi on a Pi 3 with the [Raspberry Pi 7" LCD Touchscreen](https://www.sparkfun.com/products/13733) mounted in a [SmartiPi Touch case](https://www.sparkfun.com/products/14059). I'm using the [TouchUI plugin](http://plugins.octoprint.org/plugins/touchui/), set up with [boot-to-browser](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)). Touching the screen to wake up the screen saver is problematic, because the touch counts as a click... better hope that an unwanted control isn't under the spot where you touch!

To disable the screen saver, I edited /home/pi/TouchUI-autostart/chromium.xinit and added these three lines immediately before chromium is launched:

```
xset s off
xset -dpms
xset s noblank
```

Just for good measure, I also changed these two settings in /etc/kbd/config:

```
BLANK_TIME=0
BLANK_DPMS=off
```

I'm not sure whether this should be considered an OctoPi issue or a TouchUI one.

intrglctcrevfnk, 2017-07-07:
@NF6X 
Where exactly did you place the xset arguments from your post above? Did you insert it into the  `CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG`

Or did you place it just before the script launches chromium as su?
`su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"`

Thanks for the help!  I've accidentally paused and even cancelled an octoprint job when trying to wake the screen back up.  I've tried the /etc/kbd/config edit and also used the LDXE guide to preventing sleep. Made a backup of chromium.xinit just in case.

NF6X, 2017-07-07:
I placed the xset commands immediately before the script launches chromium with su.

I'm lucky that I haven't broken anything when waking up the screen. It would have been a matter of time before that happened if I hadn't figured out how to turn off the screen saver.

intrglctcrevfnk, 2017-07-07:
Awesome!  Just one last thing (as I am not a programmer, just a hobbyist that knows enough to be dangerous).  did you use the & argument/condition/whatever between those xset lines?

so no &:
`matchbox-window-manager &
unclutter -idle 0.1 &
xset s off
xset -dpms
xset s noblank
su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"`

Or with the &:
`# Start window manager, unclutter to remove cursor and start chromium as specific user                          
matchbox-window-manager &
unclutter -idle 0.1 &
xset s off &
xset -dpms &
xset s noblank &
su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"`

My "just enough to be dangerous" thinking is that is a conditional argument?

NF6X, 2017-07-07:
Like this:

```
# Start window manager, unclutter to remove cursor and start chromium as specific user
matchbox-window-manager &
unclutter -idle 0.1 &
xset s off
xset -dpms
xset s noblank
su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"
```

intrglctcrevfnk, 2017-07-07:
Thanks again!

I ended up using the & after each line before I got a reply from you(since I had a backup why not) and it worked fine for me.  Been on for over an hour and actually seems a little brighter!  I think the dimming was part of the DPMS system.

NF6X, 2017-07-08:
You should remove the '&' signs at the ends of the 'xset' lines. xset is not a daemon which should be run as a background task; let each xset invocation exit before invoking it again.

guysoft, 2017-07-23:
Any news on this? otherwise will close due to inactivity.

NF6X, 2017-07-23:
I believe that this issue remains when using any local display, particularly a touchscreen that doesn't offer a way to un-blank the screen without blindly clicking something. I would like to see a more user-friendly way to disable the local screen saver when using OctoPi with a local display. I think it can be debated whether this would be better done in OctoPi itself, vs. a plugin such as TouchUI.

Are there an plans for the main OctoPi distribution to offer local display support without any user-installed plugins? Or would that be out of scope for the main OctoPi distribution?

BillyBlaze, 2017-11-27:
I have updated the start-up files a while ago with an option to enable and disable the screensaver in your default file (e.g. /etc/default/touchui)

Config: https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/touchui.default#L32
Actual code: https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/chromium.xinit#L48

Lepterois, 2017-12-12:
Hi all!
Thanks for this replies! Sorry I was out the subject for a while!
I try all of this as soon as I 've 5 minutes!
thanks for all!

Lepterois, 2018-01-11:
Hi All!
@BillyBlaze I tried to change the option in touchui.default but it was not working (do anything)
then I tried to change the condition in chromium.xinit (# Check if we need to disable the screensaver
if [ "$DISABLE_SCREENSAVER" = true ]) I put false instead true and it looks working (the screen stay light on). It's not very clean for a  method but it's working... Anyway thanks for all!

eboston, 2018-10-19:
So I have the DISABLE_SCREENSAVER=true in my /etc/default/touchui file.  For the most part, the screen doesn't blank.  However, every so often, it will blank.  The problem seems to be associated with booting since when this happens, if I reboot the Pi, the disable screensaver works again.