guysoft/OctoPi #365: OctoPi 0.14 daily 08 Jun 2017: Genius WideCam F100 does not work.

tubeme, 2017-06-12:
Hello,

I'm installing a Genius WideCam F100 to the OctoPi daily 0.14 and the camera does not want to stream.

The system is Pi Zero + Powered USB HUB (Zero4U UUGear) for the camera. 

The camera is able to connect to the Pi i see activity on the USB led. No streaming occurring. 

I used a Logitech UltraVision for check if the USB is OK. And it works perfect. The Logitech UltraVision is working with this HUB. 

Any Ideas?

tubeme, 2017-06-12:
I was able to run it from the third USB port. Somehow it started working. But I have to load first time the camera view with black screen and then reload it to have a picture.

tubeme, 2017-06-13:
I can initialize the camera only if I use 30FPS in 1920x1080. There is no other option to see the image. so the webcams-known-to-work should be updated. Or there is another way of make it work which I dont know...

mdjurfeldt, 2017-07-07:
I had a similar problem, but in /boot/octoprint.txt, I spotted this:
> Apparently there a some devices out there that with the current 
> mjpg_streamer release do not support the -f parameter (for specifying 
> the capturing framerate) and will just refuse to output an image if it 
> is supplied.

so I added:

> additional_brokenfps_usb_devices=("1e4e:0110")

where 1e4e:0110 is the vendor/product ID for my camera as output by lsusb.

This solved my problem.

ManuGithubSteam, 2017-08-12:
That is good to know! i just bought that cam. Can you post the boot/octoprint.txt as i don't have that file...
Thanks!

ManuelPeiro, 2017-08-26:
Hi,
I am in the same... First sorry for my poor English.
I have instaled OctoPi, in a Rasperry Pi 3, for admin mi 3d Printer. And i buy a webcam seeing this url: "https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work", where is supposed Genius F100 is working in OctoPi... but when i  plugged the Genius F100 wide webcam, i can read in control panel: "Webcam Stream is loading", and this no change... the webcam, is not working. But is supposed this model works with Octopi.

This webcam works with Octopi? how i can do, this webcam work?

I am neubie with Linux in general, and Rasperry in particular.

Can you help me? What i have to do?

With Astroprint, this webcam works....

Thanks

ManuGithubSteam, 2017-08-26:
Hi
I also got an F100 and it works without a hinch. I did not need to do the additional_brokenfps_usb_device hack.

I just plug it in and it works. And the quality is very good as well. Hope this helps.
Im using a pi 3 and normal raspian and then octoprint installed ontop.

ManuelPeiro, 2017-08-27:
hi....

i made work thewebcam "Genius F100"... with the help of "SAN YOUTUBE"...

This video show how.... https://youtu.be/AjS83guA9NQ

The resume of video:

1 - Find out the IP address of RASPERRY PI
2 - On a TELNET client, for example PUTTY, connect to that IP.
3 - You will be prompted for Username and Password, by default in OCTOPI are: pi | Raspberry
4 - With this you will be connected by TELNET to OCTOPI
5 - Change to the directory where the configuration file of the Webcam is:
                "Cd / root / bin $"    'I think the $ is to get permissions to edit files
6 - Edit the configuration file of the WEBCAM
                 Sudo nano webcamd
7- Search in this file "webcamd", the line of setup parameters of webcam
8- In the URL: "https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work", search the parameters of your webcam, and COPY to the config file in Raspberry "webcamd"
9- Save changes. XD
10- Restart Raspberry

In my case, this steps works.

I hope, this can be used for someone...

PR #493 (2018-02-26): Add Genius WideCam F100 to brokenfps
Genius WideCam F100 is a popular webcam for 3D-printers because of it's very wide FOV @ 120 deg.
It only supports one specific frame rate per resolution so disabling the frame rate option is the best option.
This will also fix #365 so that can be closed.
closed 2018-02-26 by guysoft