guysoft/OctoPi #367: OctoPi 0.14.0 RC2 status

guysoft, 2017-06-18:
Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/2017-06-18_2017-04-10-octopi-jessie-lite-0.14.0.zip
md5:
https://storage.googleapis.com/octoprint/2017-06-18_2017-04-10-octopi-jessie-lite-0.14.0.zip.md5

It is built against Raspbian 2017-04-10 ([Release Notes](http://downloads.raspberrypi.org/raspbian/release_notes.txt))

### Changes in the image

  * Added note to `/root/bin/webcamd` to use `octopi.txt` for configuration
  * Added note to `octopi-network.txt` for Mac OS X users to not use Textedit/properly configure Textedit
  * Added note with further instructions on usage to boot and login screen
  * Allow configuration of mulitple wifi networks via `octopi-wpa-supplicant.txt`
  * Better "service not running" error pages
  * Enabled SSH by default (we usually run headless, disabling it is not an option)
  * Updated stock `config.yaml` and init script for OctoPrint to 1.3.0+
  * Updated `install-desktop` script to only pull in needed packages
  * Fixes:
    * Prevent NTP updates from failing on RPi3 wifi (#327)
    * Prevent weird SSH issues on RPi3 (#294])
    * Prevent duplicate X-Scheme headers in haproxy (#239)
    * Workaround for an issue with recent `pip` versions and pyasn1 (#276)

* RC2 changes:
  * Service file to fix symlink for wpa_supplicant.conf #337
  * Custom webroot & disabled controls for webcam server #359

Note:
In 0.13.0 we promised new variant for more arm devices support. The has been a achieved with the new Armbian variant. With it, [all the devices listed here support OctoPi](https://www.armbian.com/download/), but since we didn't get a call for a specific device, we are not maintaining a build at the moment.

huelvayork, 2017-06-22:
Hi. I'd like to test the armbian variant. Where can I get it?

guysoft, 2017-06-22:
There is no build because there are 40+ different Armbian images. What device are you using?

huelvayork, 2017-06-22:
It's an orange pi zero.

guysoft, 2017-06-22:
Do you want to build one yourself?

huelvayork, 2017-06-22:
I can try to build it from within Ubuntu. Just not sure if the instructions are correct (download raspbian_latest?)

guysoft, 2017-06-23:
Download the that corresponds to the base image of your device.
Sinze you have an Orange Pi Zero its: https://dl.armbian.com/orangepizero/Ubuntu_xenial_default.7z
Taken from: https://www.armbian.com/orange-pi-zero/
You might need to re-package the image in to a zip so it could be read.

jctots, 2017-06-24:
Hello, I have tried installing this on RPi Zero W with Pi Camera module, and the webcam stream is still not working. Here is the output when I tried taking a raspistill:

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

guysoft, 2017-06-25:
Raspberrypi foundation released a new distro during our RC2. So there is an RC3 now at #371

closed 2017-06-25 by guysoft

mod38, 2017-06-25:
I have been using RC2 for the last few days with no problems, I'll download RC3 now.
My main comment is that I like to work with a fixed IP address so I still edit wpa_supplicant.conf and dhcpcd.conf to set that up just as I do on a normal full Raspbian.