guysoft/OctoPi #368: Octopi 0.13 boot issue - New Rpi3

martinold, 2017-06-18:
Imaged is flashed to a sd card and inserted in to RPi3. Boot freezes at line
4.189155] ---[ end kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)
.img downloaded from https://octopi.octoprint.org/

Possible cause please?

[image]

guysoft, 2017-06-18:
Are you sure its flashed correctly? Its likely an io error or something
like that.

martinold, 2017-06-18:
I downloaded. Extracted to desktop, burned image to sd using win32diskImager, also retried with Etcher directly from .zip file, uncommented and edited SSID/password, ejected card and put into pi and then powered up. I have also tried 2 different card types, the better one being a SanDisk 32gb hc1.

guysoft, 2017-06-18:
Its basically going over the frequent errors at this point:
What power supply and CABLE are you using? Does the top right of the screen show a colored box?
Are you plugging any USB devices in?

martinold, 2017-06-18:
Power supply 2.4 A, also tried a spare 2 A. Unsure of cable but tried some thick looking ones I have.
Coloured box appears and then fades when boot stops. Also 4 rasberry icons at top of screen.
I have tried booting with and without USB devices, keyboard, mouse, logitech cam.

martinold, 2017-06-18:
I have noticed .bin file has an association with notepad. Is this important?

guysoft, 2017-06-18:
Whichv.bin file?
Have you tried reflashing?

martinold, 2017-06-18:
The boot code .bin has an association. 
Yes to reflashing but association stays.
[image]

martinold, 2017-06-18:
I think all is well now. Flashed drive on a different computer, same card, same procedure but boots fine now. I am mystified but happy it boots.

closed 2017-06-18 by guysoft