guysoft/OctoPi #369: Octopi - Webcam URL not using Rotation Settings

WanaGo, 2017-06-21:
Running using OctoPi, using Octoprint 0.14 with latest 1.3.4, on a Raspberry Pi Zero W.

Have a Logitech (04bd:08c1) Quickcam Fusion camera, an old 1.3MP camera, interfaced to the PI, and that is streaming no problem. However the orientation is not the same as what is set up in the settings.

Under 'Webcam and Timelapse' I have it set to "Flip webcam horizontally" and "Flip webcam vertically", so I get 180 degree rotation. The preview is correct, it is streaming in the correct orientation. However when using the URL, /webcam/?action=stream it is back to original, and is not taking on these settings.

I have searched and searched and not found a specific answer.
I have tried adding "-rot 180" into the txt file, and it did nothing. I have a feeling this may only be relevant to the Pi Cameras. I also tried '-vf -hf' and that also did nothing.
But the settings in the server page itself do work correctly, its just the streaming URL doesn't utilise them.

Is this a bug or a known issue/limitation? 
Is there a way around it, so I can pass on a URL to a friend, and the camera is the correct way up for them?

guysoft, 2017-06-21:
@foosel i saw you made changes there. Anything related?

foosel, 2017-06-21:
Rotating/flipping in OctoPrint flips only in OctoPrint. It never had any control over what mjpg-streamer does with the image (and never will - the webcam might not be coming from mjpg-streamer, that is an OctoPi specific stock setting but not necessarily what people have, there are also IP cameras out there that people use), so what's delivered directly by that (`/webcam/?action=stream`) will always stay unrotated.

devsfan1830, 2017-07-13:
@foosel Should this be taken to mean that there is no way to rotate the stream from "/webcam/?action=stream"? I tried the -rot as well and it had no effect on both the GUI image and the stream. I was looking for help on the same issue since I set my Logitech webcam sideways to capture the full height of my print area.

foosel, 2017-07-13:
@devsfan1830 no. It just means that **OctoPrint** can rotate it when displaying it, but it can't force whatever webcam server it's receiving it from (and in OctoPi's case that mjpg-streamer) to rotate - thankfully btw, it would be pretty disturbing if that was possible. 

If you want the webcam server to already produce a rotated image, you need to tell this to the webcam server, not OctoPrint. The `-rot` parameter to mjpg-streamer should work - for the Raspberry Pi Camera that is, as far as I know it's not available for the uvc driver and hence USB cameras. See also [here](https://github.com/foosel/OctoPrint/wiki/MJPG-Streamer-configuration).

guysoft, 2017-09-10:
Can this be closed?

devsfan1830, 2017-09-10:
Yep!

closed 2017-09-10 by guysoft