guysoft/OctoPi #371: OctoPi 0.14.0 RC3 status

guysoft, 2017-06-25:
Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/2017-06-25_2017-06-21-octopi-jessie-lite-0.14.0RC3.zip
md5:
https://storage.googleapis.com/octoprint/2017-06-25_2017-06-21-octopi-jessie-lite-0.14.0RC3.zip.md5

It is built against Raspbian 2017-06-21 ([Release Notes](http://downloads.raspberrypi.org/raspbian/release_notes.txt))

### Changes in the image

  * Added note to `/root/bin/webcamd` to use `octopi.txt` for configuration
  * Added note to `octopi-network.txt` for Mac OS X users to not use Textedit/properly configure Textedit
  * Added note with further instructions on usage to boot and login screen
  * Allow configuration of mulitple wifi networks via `octopi-wpa-supplicant.txt`
  * Better "service not running" error pages
  * Enabled SSH by default (we usually run headless, disabling it is not an option)
  * Updated stock `config.yaml` and init script for OctoPrint to 1.3.0+
  * Updated `install-desktop` script to only pull in needed packages
  * Fixes:
    * Prevent NTP updates from failing on RPi3 wifi (#327)
    * Prevent weird SSH issues on RPi3 (#294])
    * Prevent duplicate X-Scheme headers in haproxy (#239)
    * Workaround for an issue with recent `pip` versions and pyasn1 (#276)

* RC3 changes:
  * Built on 2017-06-21

Note:
In 0.13.0 we promised new variant for more arm devices support. The has been a achieved with the new Armbian variant. With it, [all the devices listed here support OctoPi](https://www.armbian.com/download/), but since we didn't get a call for a specific device, we are not maintaining a build at the moment.

chippypilot, 2017-06-27:
I'm not sure where to put positive feedback. But just to let you know I'm running RC3 on a Pi Zero W with an Anet A6. It is working as expected, with several prints run through it so far.

thanks.

guysoft, 2017-06-27:
Positive feedbakc feedback goes here :)

020d, 2017-06-29:
It's really opened up the use of the printer to the rest of the family.  I'm using it quite a bit more as well. The SD shuffle should never happen to a good printer.

ImplementOfWar, 2017-07-14:
I just wish Octoprint could support more then one printer at a time under the same instance,  perhaps Octopi can add the feature to spin up instance's of Octoprint to support multiple printers easily.

020d, 2017-07-14:
@ImplementOfWar The multiple instance feature is doable with a plugin.  I forget which one but it adds a port selector at the top and let's you do just that.  Search the plugin repository.

guysoft, 2017-07-16:
@ImplementOfWar @020d please see/discuses this at: #5

guysoft, 2017-07-16:
Closing, released

closed 2017-07-16 by guysoft