guysoft/OctoPi #372: octopi.local web login stops responding

SLV2NON, 2017-06-26:
I'm getting ready to reimage and start again for the third time now... I can't figure out what is happening.

pi3
pitft 3.5"
creative webcam
robo3d R1

First few boots works great. I set everything up and can use my printer. I set up the access controls, send a print or two, can log out, reboot, etc.

Then for whatever reason, after a few hours of usage, after a logout or reboot I lose access to https://octopi.local. I can still navigate to the page and have restricted access. My webcam is still operable so I know the page is live... I can ssh into my pi3... but when I try and elevate my privileges on the web interface, nothing happens. I click login and the login box disappears but it doesn't log me in. Restarting doesn't help...

I'm not making any changes to the configuration, so what could be causing this and what might the fix be?

Thanks!
Logan

ChinookTx, 2017-06-26:
Try logging out and back in, or accessing the page from "incognito" mode.
I had issues with the page "remembering" me i.e. thinking i'm logged in,
but some of the options would not work.

SLV2NON, 2017-06-26:
Thank you soooo much! Browser caching didn't stand out as being a possibility... but that was definitely it. Changed to a private browsing tab and was able to access it. I guess I'll just not worry about saving my password.

ChinookTx, 2017-06-26:
I'd still like to know if it's a bug with octoprint's way of generating
cookies, or if it's a browser bug.  I've never had this happen with any
other web "apps".

SLV2NON, 2017-06-26:
I removed all cookies associated with octoprint and then opened a new window and tried to login. Same result.

SLV2NON, 2017-06-26:
Annoying... but at least there is a workaround.

foosel, 2017-06-29:
Most likely an OctoPrint issue, not OctoPi issue and already tracked in the linked ticket.

closed 2017-06-29 by foosel