guysoft/OctoPi #373: Git refuses to run as root user

cfcolaco, 2017-06-28:
don't know if this issue as been already reported, looked fast through issues didn't see any related. 

Seems that git on octopi refuses to run as root user. 

Issue was identified while troubleshooting issue pivpn/pivpn#278 on pivpn repo.

for example, due to pivpn scrip nature, it runs as a privileged user and uses git clone to create /etc/.pivpn which is needed for installation to proceed, which fails because root simply can't run git as root. 

## Git clone output

```
root@octopi:/home/pi# sudo git clone https://github.com/pivpn/pivpn.git /etc/.pivpn
Please run git without sudo, your regular user account is enough :)
root@octopi:/home/pi# git clone https://github.com/pivpn/pivpn.git /etc/.pivpn
Please run git without sudo, your regular user account is enough :)
```

Thank you in advance.

ChinookTx, 2017-06-29:
Just "sudo /usr/bin/git" instead of "sudo git".

cfcolaco, 2017-06-29:
@ChinookTx that's a way to do it, as there are so many others to work around it but that's not exactly a solution! it's a workaround! 

isn't it too much restrictive such policy?

guysoft, 2017-06-29:
@cfcolaco We put a limitation in place because people kept cloning OctoPrint as root and braking stuff.
Will note:
1. You are already root in your shell, so you don't even need sudo
2. You are cloning root stuff in to the home folder, you don't need root privileges there at all.

@foosel I think @ChinookTx found a way around our code :-/

cfcolaco, 2017-06-29:
@guysoft I don't think you have seen the whole picture. 

1 - Read the whole issue please, notice that i don't even use octopi and the I found this because of pivpn/pivpn/#278 , clearly I know that I don't need sudo when I am root, I just picked the output proviced by the user who opened the issue in pivpn. 

2. The clone is being done to `/etc/.pivpn` which is wat pivpn script does to install. 

I understand why you placed such limitation, and it's your code, I am no one to tell you what you should do or not, just think that's a bit too restrictive. I usualy try to provide a way around instead of just saying it is wrong, but unfortunately this time, I just can't think of a better way since I don't use OctoPi

foosel, 2017-06-29:
It's not code per see but a wrapper script earlier on the path than the regular git binary. Users who know what they are doing and are expanding what is installed on OctoPi on their own should hopefully also be able to solve this easily via a simple "which git" as root user ;)

cfcolaco, 2017-06-29:
@foosel You are right. 
guess users that want to use pivpn on their octopi should work around it. 

closing issue.

closed 2017-06-29 by cfcolaco