guysoft/OctoPi #377: [end Kernel Panic - not syncing...] after writing octopi 0.14 image on MicroSD

pboichut, 2017-07-06:
Hello,

I am trying to create a bootable MicroSD of Octopi 0.14.
I got error message at Pi boot (to see picture) :
[ end Kernel Panic - not syncing : VFS: Unable to mount root fs on unknown-block(179.2)]

octopi 0.14 image written on "SanDisk Ultra 16Gb microSDHC I ",  with Win32 Disk imager - 1.0
Boot on a Raspberry Pi3

I tried on 2 different sanDisk microSD card same error message.

A card compatibility problem ?

Thanks

[image]

guysoft, 2017-07-06:
Are you busing a 2A+ power supply with a GOOD power cable?
Did thia run 0.13 before?

pboichut, 2017-07-06:
Yes, I have a good power supply 2A+ 
and the 0.13 ran before... (no changes on the machine)
Thanks

guysoft, 2017-07-07:
Does clean Raspbian also do this?
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-06-23/

Update: Raspberypi foundation just released a new image which removes the metadata_csum, it might be that this features is what is causing the problem on your card (just an assumption, I don't have the hardware).

pboichut, 2017-07-09:
Same problem with a clean Raspbian...

I bought a new microSD card (Kingston) and with it octopi 0.14 works fine.
I think it is a problem with my sanDisk microSD cards (I tested on 4 sanDisk cards same problem)

Sorry to bother you...

closed 2017-07-09 by pboichut

guysoft, 2017-07-09:
Ok, just in case you want to use the sanDisk disks, you can try the new Raspbian released and see if that starts working (because of the metadata_csum):
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/

If so next OctoPi and nighties will also work with your Sandisks.

pboichut, 2017-07-10:
Tried, same problem...
It is my sanDisk cards !