guysoft/OctoPi #379: Octopi 0.14 Wireless communication

ynothna, 2017-07-10:
Hey everyone!

I'm having a bit of an odd problem. I was using 0.13 for months with no issue but upgraded to 0.14 a couple weeks ago. Since then, the app I use, Printoid, has been unable to connect. I eventually found out it's because my Octopi isn't reachable through the network by most of my devices when it's on wireless. My phone and several of my computers can't ping it at all, yet a few others can (including my main laptop, which is why it took a while to notice). When I connect the ethernet cable, suddenly both wifi and wired connections work perfectly fine.

I've tested this on two Raspberry Pi 3, one with a fresh install of Octopi 0.14. At this point I'm at a loss of what to look for and what to try next. 

Any ideas would be awesome!

Thanks

guysoft, 2017-07-11:
Are you pinging the host name or the IP address?

ynothna, 2017-07-11:
Both, neither of them work on most devices until it has a wired network connection, then immediately all of my devices can ping both it's wired and wireless address, connect to it properly, etc.

guysoft, 2017-07-12:
In that case there are three issues that might have happened:
1. Router problems
2. wifi driver has changed and created issues
3. A change in the OS now requires more power and both wifi and the CPU demands together are two high/cause overheat.

not sure how to check for each

ynothna, 2017-07-12:
My guess is on the driver change, or some other software issue. Our setup is in our IT shop so the router is taken care of and we have new devices connecting to it all the time. I don't think it should be overheating unless it's by a large amount, it happens as soon as the device is booted and happens exactly the same on two different Pi's, every time. Overloading is possible, but I haven't been able to test that.

I test the ping using the Fing app on my Android phone, in case anyone else was able to test on the newest version. Trying to ping from any of our computers as well didn't work.

guysoft, 2017-07-15:
Hey,
I am getting a similar issue on a Raspberrypi 2 and Raspberrypi  1 with RTL8188CUS wifi devices. I am getting a huge amount of packet loss and no DHCP packets

guysoft, 2017-10-20:
Closing if no reply

closed 2017-11-01 by guysoft