guysoft/OctoPi #380: Problem with free space

leobel96, 2017-07-11:
Hello, I installed Octopi on a 2gb micro sd some weeks ago and everything worked like a charm. Today I decided to transfer everything on a larger micro sd card (8 gb) because of I thought that, after updates and new plugins, 400 free mbits are too little. I used [this](https://raspberrypibeginner.wordpress.com/2013/03/25/moving-the-pis-system-to-a-larger-sdcard/) guide and everything seemed to be ok. I switched on raspberry with the new sd card and everything was ok, I expanded the file system and, then, I rebooted but, after the login in the OctoPrintUI I noted that the free space in sdcard remained 462mb. How is it possible? I attach the screenshoot. Thank you and congratulation for you work!!
[image]

guysoft, 2017-07-11:
What do you get if you run ``df -h`` in the commandline of the raspberrypi?

leobel96, 2017-07-11:
Thank you for your reply. With the new sdcard I obtain this (http://imgur.com/a/P73ai)

guysoft, 2017-07-11:
Ok, first it looks like it did expand the system.
1. Did you reboot?
2. @fossel is there a known issue?

leobel96, 2017-07-11:
Yes, I rebooted 5 times. I didn't know if post the issue here or on octoprint because is something on the borderline.

Salandora, 2017-07-11:
Can you try a Shift+F5 reload on your browser, it could be a caching problem maybe.

leobel96, 2017-07-11:
@Salandora I love you. I deleted the cache from Firefox settings and now free space is 5.6GB. Thank you so much!!!

closed 2017-07-11 by leobel96