guysoft/OctoPi #382: M600 Timeout, Failed Print

screwyluie, 2017-07-13:
So I left a print running over night as I often do but this one had a color change in it (M600). I figured the beep from the machine would wake me up eventually and I'd go change the color. All went well until I finished the color change... it moved the printhead back to the last point and just sat there doing nothing but oozing on the part. I checked octopi and it had a connection error, it had timed out.

Is this normal? I don't know how long it took me to wake up and do the color change so I don't know when the error happened but it seems to me that the M600 command should not cause a timed out connection.

Penfold81, 2017-07-14:
I had this issue and to stop it happening again I changed the communication timeout to 500 seconds. You could go longer tho

screwyluie, 2017-07-14:
thanks for the idea I'll have to try that

guysoft, 2017-07-23:
This is an OctoPrint issue, not an OctoPi issue, it it still happens, please report it to the OctoPrint issue tracker at: https://github.com/foosel/OctoPrint/

screwyluie, 2017-07-23:
fair enough. I wasn't sure how divergent the two were since I haven't used octoprint.

closed 2017-07-23 by screwyluie