guysoft/OctoPi #383: IPv6 on OctoPi

agrif, 2017-07-14:
I want to direct your attention to [this OctoPrint issue](https://github.com/foosel/OctoPrint/issues/1984), which is really an OctoPi issue. I didn't open that issue, but I'm having the same problem so I'd figure I'd report it here.

My OctoPi installation won't respond over IPv6 unless I make the changes listed in the issue above. This results in a pretty lousy user experience on my IPv6 home network, where my laptop will try the v6 address first, time out, then finally connect over IPv4 and work. This is especially annoying when my browser forgets that the v6 address doesn't work, and will periodically stall while waiting for the v6 connect to timeout.

Please consider changing the default `haproxy` config to work with IPv6.

guysoft, 2017-07-15:
Can you post your full ``/etc/haproxy/haproxy.cfg``?
A pull request would also be appreciated, since I can't test this.

closed 2017-07-17 by agrif

bagong, 2020-05-25:
Closed, but I'll still add a comment on an opposing experience: the current configuration (Mai 2020) works with ipv6 - but haproxy fails to start *if ipv6 is disabled* on system level (e.b. by specifying ipv6.disabled=1 in /boot/cmdline.txt). I think it is quite common that people disable ipv6 all together - a common scenario is the use of a vpn that doesn't support ipv6 - as a consequence of which people might try to increase security or avoid unnecessary complexity by just removing ipv6 all together.

guysoft, 2020-05-25:
Hey, AFAIK ``ipv6.disabled=1`` is not the default in raspbian. There is an issue where they used to disable IPv6 but I don't think they do anymore (I can't see that in my current ``/boot/cmdline.txt``):
https://github.com/RPi-Distro/pi-gen/pull/300

bagong, 2020-05-25:
No, it's not default, but it's is a common "1-stop"-way of disabling ipv6 from ground up. Not sure if it's documented officially, but you'll find it mentioned all over the place, including places close to the cathedral. It's the analogon of specifying ipv6.disabled=1 on the grub command line in other distros. It felt like I wanted to drop this somewhere, as my headless octopi suddenly not being reachable through ssh anymore sent me on a bit of an unintended problem hunting spree. I am glad I found the the old configuration setting referenced in the first post, as it allowed me to keep ipv6 off and get my system back ;-). I am not sure this should be considered a bug, maybe a case for documentation. I can't tell if disabling ipv6 with more involved methods wouldn't trigger the haproxy startup problem too, but I wouldn't be surprised if it did - and then it might be something to worry about more. My comment is intended to be of more informational nature...

agrif, 2020-05-25:
That... might even be a haproxy bug, then. The [v4v6](https://cbonte.github.io/haproxy-dconv/2.0/configuration.html#5.1-v4v6) option should have "no effect on non-ipv6 sockets."

It's possible I used it wrong, way back when? @bagong, can you try modifying the lines to

~~~~
frontend public
        bind *:80 v4v6
        bind *:443 v4v6 ssl crt /etc/ssl/snakeoil.pem
~~~~

and see if those work?

As an aside -- the usual way to not use IPv6 on a machine is for it to not have a routable v6 address, which usually happens automatically and is sufficient. I've had problems with Debian being cranky in the past when it can't put a name to `::1`, so I'm curious what side effects disabling v6 entirely would have.

bagong, 2020-05-26:
That worked! ;-)

agrif, 2020-05-26:
Hm. Well, the HAProxy docs seemed to indicate this would work for both v4 and v6, but I just tried it on my OctoPi install and -- it doesn't. It doesn't look like there is any way to configure HAProxy to both work with IPv6 networks and `ipv6.disabled=1` at the same time. Oh well, worth a shot!

bagong, 2020-05-26:
Okay. I'll keep that in mind if an update occurs. Thanks for looking into it! (I have multiple raspi's with the same base configuration, so I prefer to keep it my way for now ;) Might switch if I understand better)

guysoft, 2020-05-27:
@agrif Should we PR this in to the repo?

agrif, 2020-05-27:
@guysoft no, I thought it would work in both situations but after testing, it’s IPv4 only.