guysoft/OctoPi #386: Cant connect to octopi server

Arnigon, 2017-07-18:
Hi there, I used the octopi server on several occasions already and changed out my router.
Its a connect box by unity media.

Since my octopi os crashed on the sd card, I flashed the newest OS on the SD card, and typed in my w-lan configurations into the txt file.

All good, till now.
Octopi.local doesnt work, never did tho.

No problem, next step was looking up the IP in my router and I should be able to connect right to that IP.

The IP and octopi DO show up in the router, but I just cant seem to connect to it in any way.
Even pinging the server does not work at all.

[image]
This is the IP, did everything I could, Im stuck. The /Number is new with this router and I dont know what that number represents, but its not part of the direct IP, since I pinged my printer without the /Number and it did work.

Pinging does not work, connecting via SSH does not work, trying to connect via browser (octopi.local nor the IP) does not work.

Im slowly running out of ideas.

guysoft, 2017-07-19:
Can you try connecting with ethernet and test? I am not sure how to debug this.

Arnigon, 2017-07-19:
Thanks for the reply, I resolved the problem.
First off, my new router does have some problems, but for now everything runs.

I designated a static IP to the octopi (then the device wouldnt show up as unknown, but as octopi) and I could ping it.
But a few pings got through and other ones failed so I moved the PI closer to my router (a few rooms and floors to be exact) and suddenly it worked like a charm.
After having the first connection and boot from the OS, I moved the octopi back to its original place with weaker wlan signal and now it works just fine.

I guess for the initial boot, you need a pretty hefty/strong wlan signal, after that, you´re fine with a weaker one.

Original place : 2 - 11 mbit/s
Closer to the router: around 60 mbit/s

Well, now it works just fine.

closed 2017-07-19 by guysoft