guysoft/OctoPi #387: Wrong Path in Sudoers File

Sunoo, 2017-07-19:
In OctoPi, the sudoers file /etc/sudoers.d/octoprint-service has the wrong path to the 'service' command. It lists it as /sbin/service, but the correct path is /usr/sbin/service.

guysoft, 2017-07-19:
@foosel 
Looks like he is right:
```
root@octopi:/home/pi# which service
/usr/sbin/service
```
.
Its added here:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/start_chroot_script#L136

Fixing

closed 2017-07-19 by guysoft

foosel, 2017-07-19:
I know, I already stated that in #345, but it doesn't matter as long as OctoPi sticks to that "let's just never ask a password from the user for sudo anyway" approach that Raspbian ships with by default.

guysoft, 2017-07-19:
Oh hell, your solution is better.
I think we can try it in devel after CustomPiOS merge becomes stable

commit c01d0015 (2017-07-19): Fix #387