guysoft/OctoPi #388: Can't build OctoPi with Mac OS X

aviatorken89, 2017-07-19:
Hello, ever since the file structure changed with the latest devel commits I am not able to build OctoPi anymore, and the instructions are pretty vague in the README.rst file and they don't work for me using the latest Mac OS X Sierra 10.12.5.

I remember getting it setup initially was quite a chore even before the changes, but I think it needs to be made easier to make the OctoPi img file such as double clicking on a .sh file or revising the readme file to give more clear instructions for Windows and OS X alike.

I would greatly appreciate help on this. :)

Thank you!

guysoft, 2017-07-19:
I didn't know you could build it using a mac. Good to know!
Since I did't test it for mac, its likely I changed something that works differently.

Can you post what you are trying to do, and what error you aarw getting?
I need some more details how you are actually building with mac to support it.
Output with errors would help!

aviatorken89, 2017-07-19:
I don't remember how I got it work originally, since I had help from an amazon developer that lives across the country. I just remember having to use vagrant and finding the suitable Mac alternatives for apt-get and mod-probe nfs, etc that was mentioned in the readme.

It would be super helpful if we can figure out a way to make clickable script files that create the OctoPi img automagically for the large platforms: Mac, Windows.

Thank you for your work creating OctoPi!

aviatorken89, 2017-07-19:
I was able to find the conversation my friend and I had on how we got it working originally.

1) Download & install Vagrant: https://www.vagrantup.com/downloads.html

2) In the OS X Terminal: 'sudo vagrant plugin install vagrant-share --plugin-version 1.1.8'

3) 'sudo vagrant plugin install vagrant-nfs_guest'

4) 'cd OctoPi/src/vagrant'

5) 'sudo vagrant up'

And step 5 automatically built the .img file, which was removed in the commit: [https://github.com/guysoft/OctoPi/commit/bacf064f03a903149501046bb56411d3214451ce](https://github.com/guysoft/OctoPi/commit/bacf064f03a903149501046bb56411d3214451ce)

guysoft, 2017-07-19:
Ok.
So:
1. You need to run ``../../CustomPiOS/src/update-custompios-paths`` as stated in the readme. In case you didn't you need to git clone CustomPiOS too now, [see the README](https://github.com/guysoft/OctoPi#build-octopi-from-within-octopi--raspbian--debian--ubuntu).
2. I removed the build command from the provision command, because its two seprate parts, so you need to use now ``vagrant/run_vagrant_build.sh`` after you run ``sudo vagrant up``.

Hope that helps, post if you have any issues.
It seems like its just you didn't see the new build steps that were added.

aviatorken89, 2017-07-19:
I have no idea how to setup PiOS. It was nice that everything was within one repo before. If we can get to the point of avoiding the terminal and making an executable file you click on to build the images we are golden.

guysoft, 2017-07-19:
Its CustomPiOS. The reason is its a separate repo is because its a complete toolkit to build OSes now.
I'll update the readme to include it also to say so for vagrant.

All that script does is add the path of CustomPiOS in ``custompios_path``, which you can see in your vagrant its complaining about.

aviatorken89, 2017-07-19:
Ok I got farther this time, and now this error issue:

> The following SSH command responded with a non-zero exit status.
> Vagrant assumes that this means the command failed!
> 
> mount -o vers=3,udp 192.168.55.1:/Users/Ken/CustomPiOS/src /CustomPiOS
> 
> Stdout from the command:
> 
> 
> 
> Stderr from the command:
> 
> mount.nfs: requested NFS version or transport protocol is not supported

guysoft, 2017-07-19:
Ok, so good news and bad news.
The good new is your a not alone, its a common OS X error and you can see it here:
https://github.com/mitchellh/vagrant/issues/7646

The bad news is that there are several possible causes. After reading it the mains ones that stand out are:
1. Someone writes: "I had 127.0.0.1 localhost commented out in my /etc/hosts file on my mac and I uncommented it and now nfs mounting works." - two users at the end report this fixes the issue for them.
1. Try rebooting your NFS might be stuck
3. Try figuring out if you have NFS 3 or 4. If its 4 you might need to update the Vagrantfile to include a 4 and not a 3.

aviatorken89, 2017-07-19:
Looks like there was an extraneous newline character added to the `custompios_path` file which generated the above error, before you replied I typed this comment. I didn't experience what you described above, but I can confirm the newline character is what caused the previous issue I posted.

However, I have ran into a new issue: 

> ==> default: Setting up realpath (8.26-3ubuntu3) ...
> ==> default: Setting up zip (3.0-11) ...
> ==> default: Processing triggers for systemd (232-21ubuntu5) ...
> ==> default: Setting up qemu-user-static (1:2.8+dfsg-3ubuntu2.3) ...
> ==> default: Processing triggers for man-db (2.7.6.1-2) ...
> Kens-MacBook-Pro:vagrant Ken$ run_vagrant_build
> -bash: run_vagrant_build: command not found

guysoft, 2017-07-19:
try ``./run_vagrant_build.sh``
Note it has a ``.sh`` and that some POSIX distros need the ``./`` at the beginning.

aviatorken89, 2017-07-19:
That worked, it appears to be building now. Crosses fingers.

aviatorken89, 2017-07-19:
I'm assuming I can make all of the above into a seperate .sh file I can just click on whenever I want to build an image.

guysoft, 2017-07-19:
Sure.
Script what ever you want and share it :)
We are using Jenkins for continuous builds

aviatorken89, 2017-07-19:
What about that newline character I mentioned, can this be patched in a new commit for OctoPi?

closed 2017-07-19 by aviatorken89

aviatorken89, 2017-07-20:
Update: nevermind about that newline character, It wasn't an issue at all. :)