guysoft/OctoPi #391: SPI interface disabled?

spreedated, 2017-07-24:
Hi @ll,

i couldn't find any information about the SPI interface and octopi, is it somehow disabled generally?

What I try to do is controlling a WS2801 RGB-LED stripe with the same raspi i'm using to print with.
Actually it's no big deal and i'm runnig these WS2801 on several other raspis without issues.

So, are the GPIO-SPI-pins(10,9,11 or board 19,21,23) disabled for some reason?

I'm using the Adafruit-WS2801 python library, no errors, but no light.

guysoft, 2017-07-24:
Its just like in normal Raspbian.
A few years ago [I wrote this answer on Rpi stack exchange](https://raspberrypi.stackexchange.com/questions/7771/how-to-get-an-spi-analog-to-digital-mcp3204-working-with-the-gpio), it says you need to remove the blacklisted module in vanilla raspbian. I guess it still relevant.

guysoft, 2017-07-30:
Closing due to inactivity

closed 2017-10-13 by guysoft