guysoft/OctoPi #392: MP Mini Select V2 will not connect to OctoPrint on Pi 3

gspitman, 2017-07-30:
I have tried multiple restarts and cable moves, I can not get the printer to talk to Octopi.  My V1 was talking pretty well, but it died and Amazon sent me a V2 in its place.  Now I can't get it to talk over USB.

My Pi is Ethernet connected, and I have no trouble accessing via SSH or through the web.  It just won't connect to the Printer via USB.

guysoft, 2017-07-30:
1. What does ``lsusb`` show?
2. Can you see a device to connect to in the dropdown?

gspitman, 2017-07-30:
SSH is refusing the connection, so I'm not sure on lsusb. It is showing an option in the drop down though.

guysoft, 2017-07-30:
If is not working then something else is wrong

gspitman, 2017-08-01:
I agree, there were more serious issues after I posted this.  My
replacement comes tomorrow.

daviscodesbugs, 2017-09-21:
Having this same issue. I'm able to select `/dev/ttyACM0` for the serial port, and manually set a variety of baudrates. It attempts to connect for a while before timing out with the following output:

```
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
```

The results of `lsusb` are:
```
Bus 001 Device 004: ID cfce:0300
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

guysoft, 2017-09-21:
@foosel might this be an OctoPrint issue, not OctoPi?

foosel, 2017-09-22:
OctoPi not seeing the printer in OP's case: special driver needed or printer/cable to printer broken. Printer port visible but firmware doesn't respond to handshake attempts: printer firmware or connection to printer at fault.

mcgovernr, 2018-03-21:
I wanted to give this a bump because I'm having the same issue with my Monoprice Mini V2 printer. 

I do not have any communication issues with pronterface. I can connect, move axis, print, and other. 

However, I do have issues with octoprint connecting. Octoprint sends the handshake, but there is no response from the printer.

foosel, 2018-03-22:
@mcgovernr best get in touch on the [community forum](https://discourse.octoprint.org). OctoPrint not connecting to your printer isn't anything that @guysoft as the author of the underlying operating system image can do anything about.

guysoft, 2018-03-22:
BTW, does this mean I should close this issue?

foosel, 2018-03-22:
IMHO yes. But it's your tracker ;)

gspitman, 2018-03-22:
@mcgovernr I believe there was a firmware fix for this on the printer.  I
moved on to a different model so I can't confirm that it worked... but
that's where I'd be searching if I were you.

guysoft, 2018-03-22:
Ok, so it works

closed 2018-03-22 by guysoft

mcgovernr, 2018-03-22:
Thank you for the response. 

I am on the latest firmware fix. 

I opened the issue because the printer works with other hardware but not the Pi 3. Did I open this in the correct place,

Thank you for your insights 

Ryan