guysoft/OctoPi #394: Can't SSH into a clean OctoPi/OctoPrint install.

aviatorken89, 2017-08-03:
Using the latest OctoPrint devel commit and latest CustomPiOS and latest OctoPi devel branches, the connection on port 22 is refused when this previously worked on a clean install of OctoPi.

Wasn't sure if this is an OctoPrint, CustomPiOS, or OctoPi issue so thought I'd put it here to start. Hoping you can help me resolve this

`ssh pi@10.0.5.200`

guysoft, 2017-08-03:
Is this a disto you built yourself or did you use an official release?

aviatorken89, 2017-08-03:
At first I used my custom OctoPrint build, but then I tried a vanilla version just to make sure it wasn't something I messed up and still yielded the same results.

guysoft, 2017-08-03:
Did you build OctoPi or Octoprint? I am aasking about the distribution

aviatorken89, 2017-08-03:
I built OctoPi which includes latest devil branch OctoPrint in the config file.

guysoft, 2017-08-03:
Can you provide the build.log from your build?

aviatorken89, 2017-08-03:
I checked in the build log and `BASE_SSH_ENABLE=yes`

[build.log.txt](https://github.com/guysoft/OctoPi/files/1198784/build.log.txt)

guysoft, 2017-08-04:
Fixed, Please pull latest CustomPiOS.
See commit: https://github.com/guysoft/CustomPiOS/commit/f396bdbc957f3266fb3a9dc594c167e543f2f44f

aviatorken89, 2017-08-04:
Sorry but that didn't fix it either, I'm still getting the same error.

Thank you for the help!

guysoft, 2017-08-04:
Running a build to test it here also

guysoft, 2017-08-04:
Q

aviatorken89, 2017-08-04:
I'm also rebuilding one more time just to make sure

aviatorken89, 2017-08-04:
Does it have anything to do with raspi not allowing ssh unless you put an empty filed titled 'ssh' into the boot partition of the image? https://raspberrypi.stackexchange.com/a/59515/7449

aviatorken89, 2017-08-04:
Looking at previous OctoPi release image, the ssh file was not present so that can't be the problem either. :(

aviatorken89, 2017-08-04:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=176309

aviatorken89, 2017-08-04:
Issue solved but not really an ideal solution. I removed SD card from PI, put it in my mac and in the Terminal:

> cd /Volumes/boot
> sudo touch ssh

eject the SD card and put back in the Pi and it works now.

How can this be fixed though so I don't have to do this every time I want to build a new OctoPi image?

guysoft, 2017-08-04:
Think I got it now, building to test

aviatorken89, 2017-08-04:
That fixed it, thanks again! :)

closed 2017-08-04 by aviatorken89

hansiclic, 2017-09-19:
Hi, i'm having a similar problem, that prohibits me to connect to my raspberry pi 3 with a clean octopi install via ssh. 

Whenever I try to reach it, i get the following message:

`pi@raspberrypi:~ $ ssh octopi.local`
`ssh: Could not resolve hostname octopi.local: Name or service not known`

`pi@raspberrypi:~ $ ssh pi@10.170.90.191`
`Connection closed by 10.170.90.191 port 22`

I've been trying to implement your fix (https://github.com/guysoft/CustomPiOS/commit/1bf73a6c5adc0216e4a9bedfa9d6d938efc350a8) but due to my limited knowledge I am not able to make it run. I also tried to insert the "empty ssh file method" but that also didn't work

I would appreciate your help to resolve that problem!
Thank you in advance

hans

infernocs, 2017-10-17:
I have exactly the same problem, put the ssh file on both partitions of the sd still nothing.

joschrigon, 2018-01-04:
yes. same problem, but solved it.
i installed the latest 0.14.0 OctoPi Image on my tiny Pi Zero W. After successful installation WLAN is up and running, Click on the "settings" icon. Then select "Server".

On the right hand side you see the commands to shutdown, reboot and so on..
replace e.g. the reboot command by 
`sudo apt-get install ssh` 
then save and click on "system" and "reboot"
in the next step replace the reboot code by
`sudo /etc/init.d/ssh start`
ssh is now installed and started. You can connect to the Raspi now!
To logi in, the standard Rasbian login will work "pi" and "raspberry"
To enable ssh at every boot, you can use many ways. The "raspi-config" tool, or the ssh commandline:
`sudo update-rc.d ssh defaults`

That's it. Don't forget to change back the reboot command!
`sudo shutdown -r now`

good luck!

John-Mc, 2018-01-05:
@hansiclic  i believe part of your problem is that you are using the wrong command. If your left the default username (pi) and the default hostname (octopi), you should have entered `ssh pi@octopi.local`  (but you actually entered `ssh octopi.local`)

The plain `octopi.local` is what you would use as the URL to access the OctoPrint web interface from a browser, not to ssh into OctoPi.

I have another question for you: the prompt before you entered `ssh octopi.local` shows as `pi@raspberrypi:~ $`  This indicates that you are already logged on to a pi. Are you ssh'ing into OctoPi from another Raspberry Pi?

FabioBatSilva, 2018-03-02:
Same here..
This fixed the issue for me:

```
sudo apt-get install ssh

sudo rm /etc/ssh/ssh_host_*
sudo dpkg-reconfigure openssh-server

sudo /etc/init.d/ssh stop
sudo /etc/init.d/ssh start
```

SquirrelAssassin, 2018-11-10:
FabioBatSilva, thanks a million!  I also had to remove the first install, 
step1# 
apt-get remove ssh

guysoft, 2018-11-11:
Since this issue, there is a fix to regenerate ssh keys on fail. If your keys fail please post the output of:
``/var/log/regenerate_ssh_host_keys.log``
Related to: https://github.com/guysoft/OctoPi/issues/424

oldschool1964, 2020-04-16:
I just installed a clean copy of 0.17.0 (latest as of 4/15/2020) and I'm getting the same issue. Putty using SSH gives me "connection refused". Using both the IP and octopi.local in Putty, as well as ssh in the cmd on Win10, I get the connection refused. Even if I ssh in the cmd with ssh pi@octopi.local, I'll still get "connection refused".

VinnyCordeiro, 2020-04-16:
> 
> 
> I just installed a clean copy of 0.17.0 (latest as of 4/15/2020) and I'm getting the same issue. Putty using SSH gives me "connection refused". Using both the IP and octopi.local in Putty, as well as ssh in the cmd on Win10, I get the connection refused. Even if I ssh in the cmd with ssh [pi@octopi.local](mailto:pi@octopi.local), I'll still get "connection refused".

Did you activated the ssh service? Using the `sudo raspi-config` command, navigate to Advanced Options > SSH and enable it there.

rlgjr562, 2020-11-08:
I am like oldschool1964  I just installed a clean copy of 0.17of OctoPi and I cannot connect with SSH.  I see the command sudo raspi-config which I have done many times with a normal Raspberry PIs (one with a keyboard and Monitor) but how do you do that with the typical headless OctoPi.  There needs to be a way to run that or enable SSH from within the OctoPrint software

coaxial, 2021-07-17:
Same here, the console says octopi is all booted up, available at <local ip>. And yet, `ssh pi@octopi.local` or `curl http://octopi.local` fails with connection refused. Idem when using the local IP.

guysoft, 2021-07-17:
@coaxial It doesn't help posting multiple time across the issues. It only means I have to read it multiple times. If you are having a connection issue and not a but try the community forum,  not the issue tracker

coaxial, 2021-07-24:
Apologies, the GitHub UI had shown me the comment above didn't send so I thought it didn't succeed. I'll answer in the other issue.