guysoft/OctoPi #396: Octopi as WiFi Hotspot

John-Mc, 2017-08-05:
@APAeroDan was asking on another issue thread about setting up OctoPi as a WiFi Hotspot. Guysoft suggested he open a separate issue.

Here's a a solution to do that by Kenneth Jiang: [Painless WiFi for OctoPrint](url=https://medium.com/@kennethjiang/painless-wi-fi-for-octoprint-4e6b68005400)
Scroll partway down to get to the WiFi Hotspot section. This set-up will first try to connect to any known networks you have entered in to `octopi-wpa-supplicant.txt`. If it is not successful in that, it will set up it's own WiFi Hotspot.

I used this successfully with OctoPi 0.13 and with one of the release candidates for 0.14. I have not yet tried it with the stable release of OctoPi 0.14.  Note that this setup is not dependent on OctoPi. It should work with a regular Raspian installation.

guysoft, 2017-10-13:
It might be worth writing a CustomPiOS module for that now

APAeroDan, 2017-10-13:
I since have started using repetier since it has a front end that I can use
with my touchscreen.
Not a big fan of paying 60 lbs though.

John-Mc, 2018-01-31:
Update: the WiFi Hotspot linked above works fine with the OctoPi 0.14 stable release.

It does not work on OctoPi 0.15 nightly release from 2017-12-23, or any Raspian-Stretch based image that I have tried. There appears to have been a change to /etc/network/interfaces in Raspian-Stretch. Figuring that out is above my pay grade.

guysoft, 2018-03-03:
Started writing a wofi hotspot module in my spare time. Its mmoving slowly but I have some code

John-Mc, 2018-03-03:
Last I heard, @kennethjiang was thinking of updating the WiFi Hotspot referenced in the first post here to work with Raspian-Stretch and OctoPi 0.15 when he got some time. I'm not sure if he's started on that yet or not.

guysoft, 2018-03-05:
Added a wifi module that implements [this guide](http://www.raspberryconnect.com/network/item/330-raspberry-pi-auto-wifi-hotspot-switch-internet).
[See this folder in CustomPiOS](https://github.com/guysoft/CustomPiOS/tree/CustomPiOS/src/modules/auto-hotspot). Born on [this commit](https://github.com/guysoft/CustomPiOS/commit/d0c259ac9e858d158025eecd08f379bdef46bad5).

You could make this an OctoPi variant easily.

APAeroDan, 2018-03-05:
This is great guys.  I've been ignoring this lately because life, but I'm
going to dig into this again.

guysoft, 2018-03-05:
@APAeroDan I don't think this will be a default shipping, aat best a variant you could build. So what exactly would you like?
Also would like to hear @foosel if she agrees or not.

APAeroDan, 2018-03-06:
Originally, I needed the pi to act as a hotspot so I could control the pi
with my laptop without having a Network.  In the time since I have built a
local network so I may not need this feature anymore (I don't have internet
access here).  It would be a nice feature for plug and play setup though.
Just run the image and it would setup a hotspot that one could connect to
and use as is, or use the link to setup the connection to your local
network.  It could streamline the setup process.   Just a thought.

guysoft, 2018-03-06:
Actually, after looking at @foosel's work with [netconnectd](https://github.com/foosel/netconnectd) we might ship this. I think the only change should be that the wifi should not go up if the ethernet manages to connect successfully.

John-Mc, 2018-03-06:
> I think the only change should be that the wifi should not go up if the ethernet manages to connect successfully.

Isn't one of the cases that someone wants to share their Pi's ethernet internet connection via a WiFi Hotspot from the Pi?

guysoft, 2018-03-06:
@John-Mc Yes, and thats why it would be an option in the module.

guysoft, 2018-05-03:
@foosel Do you think we should add the hostpost module as nightly and see if its a good move? I can also add scripts that control it, then OctoPrint could let you set ssid and password, and enable/disable

guysoft, 2018-05-03:
@foosel we could set at /etc/suders.d/hotspot_permissions

something like this
```
pi ALL=(root) NOPASSWD: /usr/local/bin/octopi
```

That should let pi execute a single command as root without password.`

Would that help?

foosel, 2018-05-03:
Yeah, sure.

guysoft, 2019-02-11:
Ok, so there other more mature projects out, such as:
https://github.com/jasbur/RaspiWiFi

or 

https://github.com/balena-io/wifi-connect (cc @drphil3d ).

I am not as familiar with what might work best, if I push this to nightly, or make an experimental build, is this something anyone around here is willing to test?

John-Mc, 2019-02-11:
@guysoft - I'd be happy to poke around with a wifi hotspot setup to see if I can find problems. I'll be no help on the code itself, but I'm pretty good at poking around and findout out what works and what doesn't (I did a fair amount of that when Ken was writing the hotspot mentioned in the first post in this thread).

drphil3d, 2019-02-11:
Yes I would be, I’ve used WiFi-connect in the past and it was very reliable
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

drphil3d, 2019-02-12:
so when wifi-connect is initiated it can be set to use any port number specified, you should also be able to proxy octoprint through as well. If we want to use the captive portal then port 80. I have to look into the config file for wifi-connect. it has an html page that you can just add a link to in order to access octoprint.

guysoft, 2019-02-27:
Sorry this keeps getting delayed, my nightly build system is up to no good.