guysoft/OctoPi #400: OctoPrint only accessible when Cat5 plugged in.

peepsnet, 2017-08-12:
I have ver OctoPi 0.14.0
RPi3

I can only access my OctoPi web interface through eth0.

The OPi does get connected to the internet via wlan0. I have confirmed it is connected via wlan0(no cat5 cable plugged in) via connecting the OPi to my TV and using a mouse/kb to login and `ping google.com` or do an `apt-get update`. 

I have also done a `wget localhost` and I get the index.html page properly

If I plug in my eth0 cable I can access OPi via both the eth0 IP and the wlan0 IP.  When I unplug the eth0 cable I no longer have access to the OPi on my network but my OPi can still access the internet. 

Any Ideas??

Don

guysoft, 2017-08-12:
Can you please post your ``/boot/octopi-network.txt`` setting and ``/boot/octopi-wpa-supplicant.txt``?

peepsnet, 2017-08-12:
I will post those soon. 

But again the OPi does connect to my network wirelessly. I do have internet access. 

As an update... I was poking around the haproxy.cfg. I made a typo in the cfg and I got the haproxy error page which said the OPi server was not running. So, I seem to be getting to the OPi but not OctoPrint(the service was running).

peepsnet, 2017-08-12:
Success!! I figured it out. 

My home network had its own Domain Name: "home.lan"
see here at the simulator: http://tools.netgear.com/landing/gui/security/fvs336g/simulators/v_2.2.0_59.9/lan_setup.htm

I removed this, rebooted my router and it worked. 

I want to disable the octopi.local in the system. How do I do this??

peepsnet, 2017-08-12:
Seriously this is a closed issue and not a bug that needs to be addressed

guysoft, 2017-08-12:
Er - so why did you open it? I don't understand

peepsnet, 2017-08-13:
I believe this is a bug. If I have my own domain for my home network why should I have to completely disable it and lose access to all my other devices just to run OctoPrint/Pi?

It seems this is a bug and should be addressed.  The octopi.local should be able to be turned off so I can assign my own name to the device

ntoff, 2017-08-13:
Something is wrong with your setup. I have a .lan internal domain here and it works fine. 

>The octopi.local should be able to be turned off so I can assign my own name to the device

It can be, all of that is taken care of by the underlying raspbian operating system, but shouldn't be necessary, I have mine turned on and .lan still works just fine. I suggest you look through a few raspbian tutorials, and brush up on how your router's software / firmware / dns server works.

nophead, 2017-08-13:
My LAN has a .home suffix. I can access my machines more reliably with octopi.home but octopi.local mostly works as well. octopi.home is handled by the router. octopi.local needs bonjour.

LilBub, 2018-03-18:
I'd like to chime in too.
No domain setup or the like.  Straightforward DHCP setup, Netgear wifi setup.
Everything worked fine for months.  Upon powering down Pi's for printer move (printers moved 5 feet away), upon power on, none of my 3 Octorpint servers (Pi3) are coming across (checked via angry ip scanner)  Wifi signal is 100%.   I've seen a lot of message threads about Raspberry Pi wifi fails while ethernet works

Even reimaged and tried again, problem persists.

guysoft, 2018-03-18:
@LilBub 1. unrelated to the issue.
2. If you move the Pis back does itit start working again?

LilBub, 2018-03-18:
No change.  Only way they work is on ethernet now.
Wifi just seems to have failed and not sure how to get it back.  And...on all 3.
Like I said, they moved 5 feet away and directly under the transmitter/router.

guysoft, 2018-03-18:
No idea.

LilBub, 2018-03-18:
Exactly my frustration.  Worked great for months.  Now....after a power off and on...no network changes, just slight move, all 3 are wifi dead.

John-Mc, 2018-03-18:
I'd be really curious to see what happens if you install full raspian on an SD card and see if that will connect

cotecmania, 2018-03-19:
I switched to 0.15 nigthly build (stretch) an now WIFI works great without LAN cable plugged since about one week without connection problems !!!

LilBub, 2018-03-20:
What did you switch to?  Details please

cotecmania, 2018-03-22:
OctoPI Version 0.15

guysoft, 2018-03-22:
There is no version 0.15.0, there is a nightly build and a date to say which one.

guysoft, 2018-05-09:
@npbazan  This is not a bug, please post in the community forum at http://discourse.octoprint.org/

LilBub, 2018-06-29:
FYI.... resolved.
If this occurs to you/someone you know, it is because the router has been on for a long time and needs a reboot.  Mine is on a battery backup so even with power flickers, it stays on.  Powering off and on...as crazy as it sounds...fixed this.

closed 2019-07-02 by guysoft