guysoft/OctoPi #402: Debian Stretch

dylanmtaylor, 2017-08-20:
What are the plans, moving forward, to base the OctoPi image off of Raspbian Stretch?

guysoft, 2017-08-20:
There is a working nightly.  Try it out

dylanmtaylor, 2017-08-20:
Mine does not start. I couldn't ssh into pi@octopi.local, so I plugged in a monitor. Looks like there is an issue starting the HAProxy load balancer service

dylanmtaylor, 2017-08-20:
[image]

dylanmtaylor, 2017-08-20:
This is on a Raspberry Pi Zero W. The SD card is an 8GB Sandisk and the USB port has a powered hub with an Original Prusa MK2S and a Logitech V-UBM46 USB Webcam plugged in. This is the 0.15 nightly image from last night. 

[image]

guysoft, 2017-08-21:
@foosel When you are back from vacation, not sure I'll have time to find why this fails before I fly out )'( .

ntoff, 2017-08-21:
`2017-08-16-octopi-stretch-lite-0.15.img` won't connect to the wifi for me (and I also get that haproxy message).

    pi@octopi:~ $ sudo ifdown wlan0
    ifdown: interface wlan0 not configured
    pi@octopi:~ $ sudo ifup wlan0
    wpa_supplicant: /sbin/wpa_supplicant daemon failed to start
    run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
    ifup: failed to bring up wlan0-octopi
    pi@octopi:~ $

haproxy log https://gist.github.com/ntoff/3d41f1b818d7c9fcaa52e8836d7a116e

guysoft, 2017-08-21:
Wifi works for me, but also saw users on reddit complaining about stuff: https://www.reddit.com/r/raspberry_pi/comments/6ugenm/stretch_and_headless_setup_zero_w/

ntoff, 2017-08-21:
This is on a raspberry pi 3 btw, forgot about that part.

guysoft, 2017-09-10:
``octopi-wpa-supplicant.txt`` is the way to go. We kept the old one for backwards compatibility. Might remove it in later versions.

Thanks for the update about ``bind :::443 v4v6 ssl crt /etc/ssl/snakeoil.pem``. It probably means the certificate is not being generated.

ntoff, 2017-09-10:
It's not, nor is the octoprint service being started or webcamd, at least not on the images I tried to build using the build instructions.

Commenting out the line in the chroot script that attempts to disable lightdm (I think it was around 179, sysctl_if_exists lightdm disable) fixes the webcam and octoprint services not being installed (don't ask me how or why *that* works) but gencert still never runs on boot BUT it can be run manually after logging in.

guysoft, 2017-09-21:
Managed to test a bit in between. There is an issue in policykit. Its complaining about permissions.
(Writing this from a tent in the middle of the desert. So will try fix around Sunday).

guysoft, 2017-09-24:
Ok, seems like the issue is solved when installing policykit-1, testing here and should be out in the next nightly.

closed 2017-09-24 by guysoft