guysoft/OctoPi #404: Can not SSH into Octopi

hletchworth, 2017-08-25:
I have been unable to SSH in from the beginning. Whenever I had to do something directly on the Pi I would just plug in a screen and keyboard to access the command line. 

Whenever I try to ssh octopi.local or its ip address I get the following-
user$ ssh octopi.local
Connection reset by fe80::1a7b:6a10:51d4:d0a5%en1

Then if I try a few times it changes to this-
user$ ssh octopi.local
Connection closed by fe80::1a7b:6a10:51d4:d0a5%en1

And the same if I use the IP. I have tried adding 'ssh' to the SD card but that did not help.

guysoft, 2017-09-24:
You should to ``pi@octopi.local``

guysoft, 2017-09-30:
Closing if no reply

closed 2017-10-13 by guysoft

guysoft, 2017-10-24:
Might be a duplicate of #424

hletchworth, 2017-10-24:
Never figured out my issue, I just reloaded octopi. No new trouble ticket
placed.